pixcept/ioBroker.plenticore #3: Fehlermeldung "Longitude or latitude does not set. Cannot use astro."

HighControl, 2020-01-18:
Ich bekomme im ioBroker Log folgende Fehlermeldung:

plenticore.0 | 2020-01-18 10:35:28.538 | error | (19951) Longitude or latitude does not set. Cannot use astro.
plenticore.0 | 2020-01-18 10:35:28.538 | error | (19951) Longitude or latitude does not set. Cannot use astro.
plenticore.0 | 2020-01-18 10:35:28.537 | error | (19951) Longitude or latitude does not set. Cannot use astro.
plenticore.0 | 2020-01-18 10:35:28.536 | error | (19951) Longitude or latitude does not set. Cannot use astro.

Diese kommt jede Minute 4x.

StrathCole, 2020-01-18:
Schau bitte einmal in den Systemeinstellungen (ganz oben im ioBroker der Werkzeugschlüssel), ob dort bei den Haupteinstellungen die Longitude und Latitude gesetzt ist.

StrathCole, 2020-01-18:
In der aktuellsten 1.0.2 kommt die Meldung nur noch 1x beim Adapterstart.

closed 2020-01-18 by StrathCole