pixcept/ioBroker.plenticore #10: Polling interval has a min of 5000?

nobl, 2020-03-09:
I would like to poll more often than every 5 seconds but it appears adapter has a minimum of 5000ms set? At least if I set it to 3000 it uses 5000.

If this is intentional - so be it. But maybe have a tooltip added like:
`<span class="tooltiptext translate">Here you set the polling interval [min 5000, max 100000]. Default = 5000.</span>`

Also (and just a suggestion): is it reasonable to update all states although they didn't change?
Good example is network settings. States for them are updated upon every poll and they do rarely change.

Same for local.HomeBat_P for example (unless you have a kostal-supported battery).

(by the way: German is ok for discussion :) )

StrathCole, 2020-03-09:
The states are always updated, but you can see from the "lc" attribute of the state when it actually changed last time. I like to have the info about "last updated" to see if everything is "alive".

The pollTime of 5000 for now is the minimum, correct. I have not tested if it has a negative impact to allow lower settings, so I might add the tooltip in the next commits.

I'd prefer keeping discussion in English, just because I am used to it in international projects ;-)

nobl, 2020-03-09:
looking at how often senec communicates to the plenticore i can see it having no big issues with shorter intervals. but might be worth some testing .)

(not worth opening an issue (can do that if you want, though) - but question: Via modbus there is a register providing "Power Limit from EVU" as % value. Appears the api only supplies this by giving the maximum Watts value allowed (devices.local.LimitEvuAbs)?

Of course one can calculate that. Just stumbled over this because I use the % value for things.)

StrathCole, 2020-03-09:
The power limit is not available as percentage from the api. I am not sure if I could calculate it reliably.

Senec is using modbus I guess? There is a huge difference between polling the web API and modbus. I will implement a second polling time for some non-important information (e. g. once per 5 minutes) so the polling time for changing values might be lowered.

StrathCole, 2021-01-04:
The polling time minimum has been lowered now.

closed 2021-01-04 by StrathCole