pixcept/ioBroker.plenticore #9: Uncaught exception when Plenticore is down.

nobl, 2020-03-09:
If the Kostal Plenticore goes down for whatever reason (for debugging purposes a simple restart is sufficient) the adapter breaks with an uncaught exception.

Once the plenticore is available again the adapter properly restarts and situation is fixed.
But maybe you want to see if you can catch this so it handles this more gracefully.

Errorlog:

> 2020-03-09 14:19:01.567  - warn: modbus.0 (14169) Poll error count: 1 code: {"errorCode":131,"exceptionCode":4,"message":"SLAVE DEVICE FAILURE"}
> 2020-03-09 14:19:01.889  - error: plenticore.0 (17737) uncaught exception: connect ECONNREFUSED 192.168.xxx.xxx:80
> 2020-03-09 14:19:01.890  - error: plenticore.0 (17737) Error: connect ECONNREFUSED 192.168.xxx.xxx:80
> at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)
> 2020-03-09 14:19:01.895  - info: plenticore.0 (17737) [END] Stopping plenticore adapter...
> 2020-03-09 14:19:01.931  - info: plenticore.0 (17737) terminating
> 2020-03-09 14:19:01.932  - info: plenticore.0 (17737) Terminated (NO_ERROR): Without reason
> 2020-03-09 14:19:02.052  - error: plenticore.0 (17737) uncaught exception: connect ECONNREFUSED 192.168.xxx.xxx:80
> 2020-03-09 14:19:02.058  - error: plenticore.0 (17737) Error: connect ECONNREFUSED 192.168.xxx.xxx:80
> at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)
> 2020-03-09 14:19:02.460  - error: host.iobroker Caught by controller[0]: { Error: connect ECONNREFUSED 192.168.xxx.xxx:80
> 2020-03-09 14:19:02.461  - error: host.iobroker Caught by controller[0]:     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)
> 2020-03-09 14:19:02.461  - error: host.iobroker Caught by controller[0]:   errno: 'ECONNREFUSED',
> 2020-03-09 14:19:02.461  - error: host.iobroker Caught by controller[0]:   code: 'ECONNREFUSED',
> 2020-03-09 14:19:02.461  - error: host.iobroker Caught by controller[0]:   syscall: 'connect',
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[0]:   address: '192.168.xxx.xxx',
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[0]:   port: 80 }
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[1]: { Error: connect ECONNREFUSED 192.168.xxx.xxx:80
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[1]:     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[1]:   errno: 'ECONNREFUSED',
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[1]:   code: 'ECONNREFUSED',
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[1]:   syscall: 'connect',
> 2020-03-09 14:19:02.462  - error: host.iobroker Caught by controller[1]:   address: '192.168.xxx.xxx',
> 2020-03-09 14:19:02.463  - error: host.iobroker Caught by controller[1]:   port: 80 }
> 2020-03-09 14:19:02.463  - info: host.iobroker instance system.adapter.plenticore.0 terminated with code 0 (NO_ERROR)
> 2020-03-09 14:19:02.463  - info: host.iobroker Restart adapter system.adapter.plenticore.0 because enabled
> 2020-03-09 14:19:02.571  - info: modbus.0 (14169) Disconnected from slave xxx

nobl, 2020-03-09:
Have to ammend:
While the adapter appeared to be running again it stopped polling the plenticore after this until I restarted it.

Weather forecast _appeared_ to be polling still - but cannot not confirm if it truly did.

StrathCole, 2020-03-09:
This should hopefully be fixed by latest commit. Thank you.

commit 0d25d394 (2020-03-09): - added handling for failed logins and exception in Api requests, fixes #9
closed 2020-03-09 by StrathCole