pixcept/ioBroker.plenticore #12: forecast kills adapter

neorulez, 2020-04-08:
Hi,

after updating my wifi access points yesterday in the night I noticed a problem with the adapter today. I checked my grafana data this morning and noticed that there are no data for my pv.
So I had a look into iobroker itself and the log:
`host.raspberrypi | 2020-04-08 08:08:31.972 | info | Restart adapter system.adapter.plenticore.0 because enabled
-- | -- | -- | --
host.raspberrypi | 2020-04-08 08:08:31.971 | info | instance system.adapter.plenticore.0 terminated with code 0 (NO_ERROR)
host.raspberrypi | 2020-04-08 08:08:31.971 | error | Caught by controller[0]: at process._tickCallback (internal/process/next_tick.js:68:7)
host.raspberrypi | 2020-04-08 08:08:31.971 | error | Caught by controller[0]: at promise.then (/opt/iobroker/node_modules/standard-as-callback/built/index.js:19:49)
host.raspberrypi | 2020-04-08 08:08:31.971 | error | Caught by controller[0]: at tryCatcher (/opt/iobroker/node_modules/standard-as-callback/built/utils.js:11:23)
host.raspberrypi | 2020-04-08 08:08:31.970 | error | Caught by controller[0]: at client.get (/opt/iobroker/node_modules/iobroker.js-controller/lib/states/statesInRedis.js:572:33)
host.raspberrypi | 2020-04-08 08:08:31.970 | error | Caught by controller[0]: at /opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:2065:22
host.raspberrypi | 2020-04-08 08:08:31.969 | error | Caught by controller[0]: TypeError: Cannot read property 'val' of null
plenticore.0 | 2020-04-08 08:08:31.599 | info | (7036) Logged out from API
plenticore.0 | 2020-04-08 08:08:31.443 | info | (7036) Terminated (NO_ERROR): Without reason
plenticore.0 | 2020-04-08 08:08:31.442 | info | (7036) terminating
plenticore.0 | 2020-04-08 08:08:31.266 | info | (7036) [END] Stopping plenticore adapter...
plenticore.0 | 2020-04-08 08:08:31.260 | error | at process._tickCallback (internal/process/next_tick.js:68:7)
plenticore.0 | 2020-04-08 08:08:31.260 | error | at promise.then (/opt/iobroker/node_modules/standard-as-callback/built/index.js:19:49)
plenticore.0 | 2020-04-08 08:08:31.260 | error | at tryCatcher (/opt/iobroker/node_modules/standard-as-callback/built/utils.js:11:23)
plenticore.0 | 2020-04-08 08:08:31.260 | error | at client.get (/opt/iobroker/node_modules/iobroker.js-controller/lib/states/statesInRedis.js:572:33)
plenticore.0 | 2020-04-08 08:08:31.260 | error | at /opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:2065:22
plenticore.0 | 2020-04-08 08:08:31.260 | error | (7036) TypeError: Cannot read property 'val' of null
plenticore.0 | 2020-04-08 08:08:31.260 | error | (7036) uncaught exception: Cannot read property 'val' of null
plenticore.0 | 2020-04-08 08:08:30.644 | info | (7036) calcMinSoC for day 2.
plenticore.0 | 2020-04-08 08:08:29.899 | info | (7036) Using forecast of day 1 for MinSoC calculation.
plenticore.0 | 2020-04-08 08:08:29.898 | info | (7036) calcMinSoC for day 1.
plenticore.0 | 2020-04-08 08:08:29.724 | info | (7036) Read forecast data before calcMinSoC.
plenticore.0 | 2020-04-08 08:08:29.722 | info | (7036) Still 1 adapters to check.
plenticore.0 | 2020-04-08 08:08:29.719 | info | (7036) Using darksky.0 in weather forcast.
plenticore.0 | 2020-04-08 08:08:29.675 | info | (7036) Still 2 adapters to check.
plenticore.0 | 2020-04-08 08:08:29.653 | info | (7036) Still 3 adapters to check.
plenticore.0 | 2020-04-08 08:08:29.651 | info | (7036) Enabling MinSoC forecast data.
plenticore.0 | 2020-04-08 08:08:29.637 | info | (7036) Enabling forecast data.
plenticore.0 | 2020-04-08 08:08:28.533 | info | (7036) [INFO] Configured polling interval: 20000
plenticore.0 | 2020-04-08 08:08:28.531 | info | (7036) Configured Wp of panel(s) is xxxxx
plenticore.0 | 2020-04-08 08:08:28.530 | info | (7036) Got all internal forecast data and made it available to main process.
plenticore.0 | 2020-04-08 08:08:28.509 | info | (7036) Requested alternative weather data from kachelmannwetter.com
plenticore.0 | 2020-04-08 08:08:28.372 | info | (7036) Requested weather data from kachelmannwetter.com
plenticore.0 | 2020-04-08 08:08:28.210 | info | (7036) City id for coordinates xx.xxxx/yy.yyyy is yxyxyxyx
plenticore.0 | 2020-04-08 08:08:27.717 | info | (7036) got weather data from met.no with 235 elements.
plenticore.0 | 2020-04-08 08:08:26.789 | info | (7036) [START] Starting adapter plenticore v2.1.1r217

`

So I checked for updates and there was an update for the plenticore adapter from 2.1.0 to 2.1.1
JS controller has been up-to-date which I checked by:
`sudo iobroker stop
sudo iobroker update
sudo iobroker upgrade self
sudo iobroker start`

The error persisted...

Finally I deactivated the forecast option as I saw in the log that the connection to the inverter was fine so far. This made the adapter stable.

Any idea what to do in order to make forecast work again?

Best regards

StrathCole, 2020-04-08:
Do you have a battery attached?

neorulez, 2020-04-08:
Nope. Just 2 strings connected to input 1 and 2.
The adapter worked flawlessly until the update of my access points and so the loss of connection to the inverter. The connection between iobroker (btw installed on a Raspberry Pi 4) and the inverter is fine. And even the connection to kachelmannwetter.com seems to be ok - as you can see from the log. It just might be something in the forecast calculation.

StrathCole, 2020-04-08:
Okay, I had only instances with battery connected, yet. So this seems to be related to a missing value without battery. I will check that.

StrathCole, 2020-04-08:
You should now update to git version (r219) to solve the issue.

closed 2020-04-08 by StrathCole

neorulez, 2020-04-08:
Seems to be OK now.
Great work! Thx!

commit 96e3b6fc (2020-04-08): - fixed problem with missing battery, fixes #12