pixcept/ioBroker.plenticore #15: Using Plenticore Adapter only for forecast

kaybau, 2020-05-27:
Hello,
I have a Kostal BA inverter which is connected via Javascript to iobroker.
I want to use your adapter for forecast reasons, 
Is there any chance to deactivate the polling to Plenticore address?
beside of Plenticore adapter I have daswetter. Where are the datapoints stored for forecast?
under Plenticore/forecast I have 2 DP's. one is website to kachelmann, second DP is my city.

Cheers,
Kay

StrathCole, 2020-05-27:
No sorry, the polling is needed for the adapter to work. It is not planned to make a forecast-only adapter.

StrathCole, 2020-05-27:
The datapoints are at `plenticore.forecast.current/day1/day2`.

kaybau, 2020-05-27:
a pity. forecast data will only be generated, if Plenticore data are available.

closed 2020-11-05 by StrathCole