pixcept/ioBroker.plenticore #16: EnergyFlow.Yield

NielsWL, 2020-09-16:
Hey...

The data EnergyFlow.Yield is not the complete yield of the PV-Plant. The energy that is used for charching the battery is missing. Can you add data fields (day, month, year, total) so that all the yield of the plant can be logged?

StrathCole, 2020-09-16:
> The data EnergyFlow.Yield is not the complete yield of the PV-Plant.

Yes, but that is the way KOSTAL fills the yield field (consistend with KOSTAL and PIKO portal).

The Fields for year, day etc. are available, e. g. `plenticore.0.scb.statistic.EnergyFlow.YieldDay`

NielsWL, 2020-09-16:
i know...

but can you add it? I can get the data by a blookly script, but maybe you can add it in the adapter directly.

```
var PV_Ertrag1, PV_Ertrag2;
schedule("* * * * * *", function () {
  PV_Ertrag1 = getState("plenticore.0.scb.statistic.EnergyFlow.YieldDay").val;
  PV_Ertrag2 = (typeof PV_Ertrag2 == 'number' ? PV_Ertrag2 : 0) + getState("plenticore.0.devices.local.battery.Charge_P").val / 3600;
  setState("plenticore.0.scb.statistic.EnergyFlow.PVErtrag"/*PVErtrag*/, (PV_Ertrag1 + PV_Ertrag2), true);
  console.log((PV_Ertrag1 + PV_Ertrag2));
});
```

StrathCole, 2020-09-16:
The problem is that this value is not available from the inverter. I would have to calculate the value myself and it would be more inaccurate the higher the set polling interval in the instance config is.

NielsWL, 2020-09-16:
ok. then i calculate it with the blockly script. thanks.

closed 2020-11-05 by StrathCole