pixcept/ioBroker.plenticore #18: Multi-Plenticore

Schwunkel, 2020-10-31:
Wir haben zwei Plenticores hinter einem Zähler, einer davon mit Batterie. Gibt es eine Möglichkeit beide in einem Adapter anzeigen zu lassen?

StrathCole, 2020-12-05:
Leider nein. Das geht nur mit zwei Instanzen des Adapters.

menkej, 2022-11-27:
> Wir haben zwei Plenticores hinter einem Zähler, einer davon mit Batterie. Gibt es eine Möglichkeit beide in einem Adapter anzeigen zu lassen?

Nur aus Neugier: warum möchtest du das, @Schwunkel? Für einfachere vis?

Schwunkel, 2022-11-27:
> 

@menkej ich muss mich jetzt outen: ioB läuft bei mir seit 5 Jahren und ich habe keine VIS :)))

aus zwei Gründen würde ich mich über einen Adapter für beide:
1. zwei Instanzen/Adapter heißt doppelter RAM-Bedarf -> ich hätte 80-100 MB mehr zur Verfügung
2. ich rechne mit den Daten aus beiden Instanzen -> dadurch, dass sie nicht synchron abgerufen werden kommt es zu ein paar kleineren Problemen/Abweichungen die ich mir sparen könnte.

Grüße