pixcept/ioBroker.plenticore #21: Adapter startet nicht, wenn keine Ortsangaben im iobroker eingetragen sind

KoMa1978, 2020-12-05:
Hallo,

ich nutze die Prognose nicht, daher hab ich im Plenticore Adapter auch nichts konfiguriert.

Im iobroker selber habe ich auch nichts konfiguriert, da ich die Prognose-Funktion nicht nutze/nutzen möchte.

Der Adapter lässt sich aber so nicht starte und bricht mit der Fehlermeldung "error	(21200) Astro functions not available as system's longitude and latitude were not found. Please check ioBroker global system config." ab.

VG,
 KoMa

StrathCole, 2020-12-05:
Die Standortdaten werden vom Adapter immer benötigt, da er diese nicht nur für die Prognose nutzt sondern auch für die aktuelle "potenzielle" Leistung.

KoMa1978, 2020-12-05:
OK ...

aber fällt nicht "potenzielle Leistung" auch irgendwie unter Prognose?

Und bevor der Adapter dann gar nicht startet, wäre es nicht besser, eine Warnung/Info im Log auszugeben und einen Fallback auf "Tag = 09:00-18:00" besser? Oder zumindest eine kurze Notiz im Readme, damit man nicht erst suchen muss, warum der Adapter nicht startet.

Besten Dank,

StrathCole, 2020-12-05:
Ja, da gibt es sicher Verbesserungspotenzial. Das Problem ist, dass die genutzten Wetteradapter derzeit nicht deaktiviert sind und ohne Ortsangaben diese Fehler werfen. Daher wäre es eine größere Umstellung. Werde ich mich bestimmt drum kümmern, ist nur gerade nicht so drin zeitlich.

KoMa1978, 2020-12-05:
Alles gut!!

Der Adapter an sich ist klasse und um Welten besser und komfortabler als eine Anbindung per modbus.
Das ist ja alles auch nicht dringend ... mein WR wurde letzte Woche installiert und nur geht's an die Anbindung an den ioBroker und das ist mir bei der Installation aufgefallen.

Wenn es was zu testen gibt, meld dich gerne.