pixcept/ioBroker.plenticore #22: Unterstützung von HTTPs

KoMa1978, 2020-12-05:
Hi,

es wäre klasse, wenn der Connect zum Wechselrichter wahlweise per HTTP oder HTTPS erfolgen könnte.

Der Einbau als Option dürfte ja relativ einfach sein, da Kostal HTTPs aber sehr stiefmütterlich behandelt, bitte unbedingt daran denken, dass ungültige Zertifikate ignoriert werden (ich vermute die Option ist "rejectUnauthorized: false").

Ansonsten läuft der Adapter hier mit einem Plenticore Plus 10 (Firmware 01.17.05075) auf anhieb und echt gut!!
Du solltest gucken, ob man den nicht in Richtung "stable" bringen kann, damit er im iobroker direkt zum Download angeboten wird.

Besten Dank,
 KoMa

StrathCole, 2020-12-05:
Ich muss mal schauen, dass ich eine https-Option noch einbinde. Danke für den Hinweis, habe ich für Kostal selbst nie genutzt.

KoMa1978, 2020-12-05:
Ich versuche so weit es geht, alle Geräte verschlüsselt anzusprechen ... Leider wird dieses Thema seeehr stiefmütterlich behandelt, nicht nur von Kostal, sondern auch von vielen anderen.

Da man leider das https-Zertifikat im WR nicht tauschen kann und das Zertifikat ungültig ist (abgelaufen, auf den Hostnamen SCB, ...) unbedingt ungültige Zertifikate erlauben.

Besten Dank auf jeden Fall schon einmal und wenn was getestet werden soll, melde Dich gerne.

StrathCole, 2021-06-09:
Ist nun in Version 2.2.0 möglich.

closed 2021-06-09 by StrathCole