pixcept/ioBroker.plenticore #23: Warnung vor neuer Firmware

Tranquility76, 2020-12-09:
Falls noch nicht gesehen - mit der neuen Firmware wandert die Batteriesteuerung in den Installateurbereich und dürfte damit den Adapter hier kaputt machen was die Batteriesteuerung betrifft! :-(

Datum (JJJJ.MM.TT): 2020.11.20
KOSTAL_Update_PLENTICORE_plus_PIKO_IQ_01.17.05075

[Release Notes](https://www.kostal-solar-electric.com/de-de/download/-/media/document-library-folder---kse/2019/03/12/07/13/release_notes_011705075.txt)

StrathCole, 2020-12-10:
Ja, das ist leider korrekt. Es geht aber nur um eine Einstellung, die ich "probehalber" in der letzten git-Version hinzugefügt hatte (externe Steuerung (de-)aktivieren).

Schwunkel, 2020-12-14:
schon getestet ob ein Downgrade möglich ist? Ich könnte KOSTAL_Update_PLENTICORE_PIKO_IQ_011605025.swu zur Verfügung stellen!

Tranquility76, 2020-12-14:
So wie es aussieht, bleiben die bisherigen Funktionen im Anlagenbetreiberbereich. Nur die Aktivierung der externen Steuerung (ModBUS) wandert in den Installateurbereich.

Das Changelog klang da etwas anders.

Von daher keine Gefahr für den Adapter ;-) und der Thread hier kann aus meiner Sicht auf „erledigt“ gesetzt werden.

closed 2020-12-15 by StrathCole

celle1234, 2021-01-03:
> schon getestet ob ein Downgrade möglich ist? Ich könnte KOSTAL_Update_PLENTICORE_PIKO_IQ_011605025.swu zur Verfügung stellen!

Hallo könntest du mir die SW zur Verfügung stellen. Bisher haben Downgrades funktioniert.

Schwunkel, 2021-01-03:
> 
> 
> > schon getestet ob ein Downgrade möglich ist? Ich könnte KOSTAL_Update_PLENTICORE_PIKO_IQ_011605025.swu zur Verfügung stellen!
> 
> Hallo könntest du mir die SW zur Verfügung stellen. Bisher haben Downgrades funktioniert.

https://drive.google.com/u/0/uc?id=18xVGb6FeUC-ANo7o6U4iG2e7PfQaqVuC&export=download

Bitte schön.

celle1234, 2021-01-04:
Danke. Das Downgrade hat funktioniert. Und auch die Steuerung darüber geht. Die ist nur scheinbar ein bischen speziell.