pixcept/ioBroker.plenticore #24: "Warning: Request Settings ..." for each polling interval

ninja-warrior, 2021-01-04:
Hey guys,

just want to let you know that I get a lot of new warnings for plenticore adapter in iobroker log since I updated the adapter too newest version 2.1.4

The warnings look all the same:

2021-01-04 10:30:17.658 - warn: plenticore.0 (20388) Requesting settings - [{"moduleid":"devices:local","processdataids":["Dc_P","DigitalIn","EM_State","HomeBat_P","HomeGrid_P","HomeOwn_P","HomePv_P","Home_P","Inverter:State","LimitEvuAbs"]},{"moduleid":"devices:local:ac","processdataids":["CosPhi","Frequency","L1_I","L1_P","L1_U","L2_I","L2_P","L2_U","L3_I","L3_P","L3_U","P","Q","S"]},{"moduleid":"devices:local:battery","processdataids":["Cycles","SoC","I","U","P"]},{"moduleid":"devices:local:pv1","processdataids":["I","U","P"]},{"moduleid":"devices:local:pv2","processdataids":["I","U","P"]},{"moduleid":"scb:export","processdataids":["PortalConActive"]},{"moduleid":"scb:statistic:EnergyFlow","processdataids":["Statistic:Autarky:Day","Statistic:Autarky:Month","Statistic:Autarky:Total","Statistic:Autarky:Year","Statistic:EnergyHome:Day","Statistic:EnergyHome:Month","Statistic:EnergyHome:Total","Statistic:EnergyHome:Year","Statistic:EnergyHomeBat:Day","Statistic:EnergyHomeBat:Month","Statistic:EnergyHomeBat:Total","Statistic:EnergyHomeBat:Year","Statistic:EnergyHomeGrid:Day","Statistic:EnergyHomeGrid:Month","Statistic:EnergyHomeGrid:Total","Statistic:EnergyHomeGrid:Year","Statistic:EnergyHomePv:Day","Statistic:EnergyHomePv:Month","Statistic:EnergyHomePv:Total","Statistic:EnergyHomePv:Year","Statistic:OwnConsumptionRate:Day","Statistic:OwnConsumptionRate:Month","Statistic:OwnConsumptionRate:Total","Statistic:OwnConsumptionRate:Year","Statistic:Yield:Day","Statistic:Yield:Month","Statistic:Yield:Total","Statistic:Yield:Year","Statistic:CO2Saving:Day","Statistic:CO2Saving:Month","Statistic:CO2Saving:Year","Statistic:CO2Saving:Total"]}]) failed with code 503: {"message":"internal communication error, try again later"}

Maybe, you can spend a minute and take a look for the reason of the new warnings.

Thanks in advance and best regards,
David

StrathCole, 2021-01-04:
Which firmware are you using in the Kostal?

ninja-warrior, 2021-01-04:
I am not sure whether this is this information you are asking for:

Typenbezeichnung PLENTICORE plus 5.5
UI-Version 01.14.04213
MC-Version 01.42
IOC-Version 01.42
HW-Version 0100

StrathCole, 2021-01-04:
This one is quite old. You might need to update the firmware to use the newest adapter version.

ninja-warrior, 2021-01-04:
Done. Everythink works like a charm. Thanks!

closed 2021-01-04 by ninja-warrior