pixcept/ioBroker.plenticore #25: Release?

Bernd1984, 2021-01-06:
Hallo,
es scheint, als würde der Adapter gut funktionieren.
Wann wird dieser denn offiziell in den ioBroker aufgenommen?
Vielen Dank & Gruß, Bernd

StrathCole, 2021-01-06:
Hi, habe aktuell das Problem, dass ich den Travis-Test nicht zum Laufen kriege. Ohne den kann der Adapter nicht eingereicht werden.

Bernd1984, 2021-01-06:
Hallo Marius, 

danke für deine Antwort.

Ich bin noch sehr neu im Thema ioBroker und habe deine ZipFile in /opt/iobroker/node_modules kopiert bzw. entpackt.

Danach kann ich ihn auch unter Adapter sehen. Leider kann ich ihn dann aber nicht installieren.

Mache ich etwas falsch?

Vielen Dank & beste Grüße, Bernd

StrathCole, 2021-01-12:
Wenn du noch ein oder zwei Tage wartest, sollte der Adapter in der offiziellen Liste auftauchen. Habe ihn im Stable-Repository eingereicht.

closed 2021-01-12 by StrathCole