pixcept/ioBroker.plenticore #31: Warnings wegen nicht existierendem Objekt

nobl, 2021-02-17:
Folgende Warnings kommen bei jedem Datenabruf:

(840) State "plenticore.0.forecast.day1.power.4h.generated" has no existing object, this might lead to an error in future versions
(840) State "plenticore.0.forecast.day1.power.4h.generated" has no  existing object, this might lead to an error in future versions
(840) State "plenticore.0.forecast.day1.power.4h.generated" has no  existing object, this might lead to an error in future versions

Dabei ist der Teil "4h" variabel. Mal 1h, 3h, ...
Der betreffende Datenpunkt ist nicht angelegt.

Stoppt man den Adapter, löscht "forecast.day1" komplett und startet den Adapter danach wieder, so wird der Datenpunkt angelegt.
Einen Tag später ist er dann wieder gelöscht und die Warnings trudeln erneut ein.

virusbrain, 2021-02-22:
Hi, bei mir kann ich das gleiche Verhalten beobachten. Wobei ich meine, dass das erst seit dem Update des JS-Controllers auf 3.2.16 so ist?

nobl, 2021-02-22:
Hat nichts mit 3.2.16 zu tun, denn meiner läuft noch auf 3.2.12.

StrathCole, 2021-03-01:
Bitte mal schauen, ob das nach dem Update auf 2.1.8 immer noch ist.

nobl, 2021-03-01:
Ich habe eben einmal aktualisiert und werde das nun beobachten.
Für den Moment ist erstmal Ruhe - aber das war auch der Fall, hatte man die Datenpunkte alle gelöscht und den Adapter neu gestartet. 

Genaueres dürfte es leider erst morgen oder übermorgen zu vermelden geben. (Im Zweifel kurz anpingen, dann prüfe ich das.)

StrathCole, 2021-03-01:
> Genaueres dürfte es leider erst morgen oder übermorgen zu vermelden geben. (Im Zweifel kurz anpingen, dann prüfe ich das.)

Danke. Ich versuche auch dran zu denken, bei mir selbst zu schauen. Leider gibt das Log nicht her, an welcher Stelle das Problem auftritt, daher ist es schwer zu debuggen.

StrathCole, 2021-03-03:
Ich schließe mal, weil es bisher nicht mehr aufgetaucht ist.

closed 2021-03-03 by StrathCole