pixcept/ioBroker.plenticore #32: Error getting forecast data from met.no api: null

Seffe-00, 2021-03-01:
Seit heute Morgen werden die Prognose Daten nicht mehr aktualisiert.

Könnte es vielleicht mit dem Update der Metno API zusammenhängen?.

Siehe

https://api.met.no/

News:
Locationforecast and Nowcast
As mentioned previously, the old versions of Locationforecast (1.9), Nowcast (0.9) and Weathericon (1.1) will terminate on March 1st, 2021. Incredibly, almost 70 % of the current API traffic is still using version 1.9, This means a lot of sites and apps will stop working shortly. We recommend everyone not yet upgraded to start working on this ASAP. Please see our docs for more information:

Gruß Seffe

StrathCole, 2021-03-01:
Danke für den Hinweis. Hatte ich gar nicht mitbekommen.

StrathCole, 2021-03-01:
Sollte mit neuer Version 2.1.8 behoben sein.

closed 2021-03-01 by StrathCole

Seffe-00, 2021-03-01:
Danke für die schnelle Reaktion.

Forecast_Current_Power:

Max wird nun aktualisiert

Jedoch die Werte:
Sky
Skyvis
Skyrain 

werden nicht aktualisiert.

Bei Rain steht bei metno noch

null mm

Kann mich nicht Erinnern ob dies schon immer so war?

Gruß Seffe