pixcept/ioBroker.plenticore #45: Can not query settings: "internal communication error, try again later"

ingmr, 2021-09-09:
My PLENTICORE plus 10 seems to not support the setting `Battery:ExternControl`. I see always warnings in the ioBroker logs that the settings request is failing. I've checked this request and the property `Battery:ExternControl` seems to cause the error.  If I remove this setting the Request succeeds. Can we somehow exclude this setting by configuration in ioBroker.

```
POST http://192.168.x.x/api/v1/settings
Content-Type: application/json
Authorization: Session *********************************
[
  {
    "moduleid": "devices:local",
    "settingids": [
      "Battery:ExternControl"
    ]
  }
]

HTTP/1.1 503 SERVICE UNAVAILABLE
Server: nginx/1.15.2
Date: Thu, 09 Sep 2021 09:01:18 GMT
Content-Type: application/json
Content-Length: 60
Connection: keep-alive
Access-Control-Allow-Origin: *

{
  "message": "internal communication error, try again later"
}
```

```
Device
Name: Kostal-Wechselrichter
Type designation: PLENTICORE plus 10
Article number: 10335959
UI version: 01.15.04581
MC version: 01.43
IOC version: 01.42
HW version: 0200
Guideline: DE VDE4105 cosphi=1
Battery input: released
```

StrathCole, 2022-04-11:
Could you please update your Plenticore firmware? This setting was introduced in a later firmware. Yours is very outdated.

ingmr, 2022-04-11:
Thanks for the hint. Firmware upgrade helps:

```
Device
Name: Kostal-Wechselrichter
Type designation: PLENTICORE plus 10
Article number: 10335959
UI version: 01.22.06892
MC version: 01.65
IOC version: 01.60
HW version: 0200
Guideline: DE VDE4105 cosphi=1
Battery input: released
```

closed 2022-04-11 by ingmr