pixcept/ioBroker.plenticore #47: Add possibility to configure the feed-in limit

PastCoder, 2022-01-16:
Currently the factor for the feed-in limit is hardcoded to 0.7. 
In some cases not 0.7 of the peak power is allowed to be fed into the grid but other percentage (e.g. 0.5 in case of some promotion for batteries).

PastCoder, 2022-01-18:
@StrathCole 
Hopefully I am doing thing right in Github in the second attempt....
And doing step by step. I did reset the fork to the initial version from your master version. 
I did the proposed changes for this change in a branch: https://github.com/PastCoder/ioBroker.plenticore/tree/configure-feed-in-limit
Please have a look there. If everything is okay, I'll create a new pull-request so that it can become part of the official version.

StrathCole, 2022-01-18:
I think you will need to create a pull request from there because I don't know how to compare the code changes otherwise.

PastCoder, 2022-01-18:
> I think you will need to create a pull request from there because I don't know how to compare the code changes otherwise.

done

StrathCole, 2022-02-01:
Is this issue done?

PastCoder, 2022-02-01:
Yes. I'll close it

closed 2022-02-01 by PastCoder