pixcept/ioBroker.plenticore #56: Fehler bei "Dynamisches Batterie Management" aktivieren?

DavidPasch, 2022-03-18:
Hier herrscht gerader purer Sonnenschein. Ortsangaben im Adapter stimmen.
Im Adapter ist "Dynamisches Batterie Management" angeschaltet.

Das hat zur Folge, das aktuell im Plenticore "intelligente Batteriesteuerung aktivieren" aktiv wird, der Akku NICHT geladen wird , dafür aber 7,3kW in Netz eingespeist werden. Prognose ist hier top, müßte der Adapter die Einstellung im Plenticore aktuell nicht deaktivieren, das primär der Akku geladen wird? Der steht bei 50% ... und wird eben nicht geladen. MAche ich das manuell, wird der Akku geladen ... was ich bei der Überproduktion auch erwartet hätte.

Bei aktiviertem "Dynamisches Batterie Management" im Adapter, habe ich allerdings auch noch NIE die Option "intelligente Batteriesteuerung aktivieren" im Plenticore deaktiviert gesehen.

Alles etwas verwirrend in der Bezeichnung. Sicher, das dies korrekt arbeitet?

In wie fern spielt "Minimaler SoC zur Aktivierung des Batteriemanagements" da noch mit rein?

NielsWL, 2022-03-18:
Ich hatte auch das Gefühl, dass die "Intelligenz" manchmal nicht so
intelligent ist und habe vor ein paar Tagen folgendes programmiert. Ich
hatte vorher häufig beobachtet, dass die Batterie zu spät anfängt zu laden
und dann doch nicht voll wird.

[image: image.png]
[image: image.png]

DavidPasch, 2022-03-18:
> Ich hatte auch das Gefühl, dass die "Intelligenz" manchmal nicht so intelligent ist und habe vor ein paar Tagen folgendes programmiert. Ich hatte vorher häufig beobachtet, dass die Batterie zu spät [...]

Deine Screenshot sind nicht mitgesendet?

NielsWL, 2022-03-18:
Eigentlich schon... Zumindest in meiner Mail.

NielsWL, 2022-03-18:
[image]

[image]

NielsWL, 2022-03-18:
Heutiges Ergebnis meiner Programmierung.

[image]

StrathCole, 2022-03-18:
Die "intelligente Batteriesteuerung" ist eine Einstellung im KOSTAL, nicht im Adapter. Der Adapter hat diese normalerweise aktiviert, er deaktiviert sie nur, wenn die Prognose für den (restlichen) Tag schlecht ist und für die Ladung der Batterie vermutlich nicht ausreicht. Im KOSTAL bedeutet das **deaktivieren** der intelligenten Steuerung, dass die Batterie geladen wird. Die **aktive** Steuerung bedeutet, dass die Ladung zurückgehalten wird, bis entweder die Abregelung erreicht ist, oder der WR "denkt" (und das macht er in der Regel schlecht), dass er jetzt laden sollte.

Im Adapter kann ich da nicht wirklich was machen, bei mir funktioniert es gut. Ist die Kapazität der Batterie bei euch korrekt in der Konfig des Adapters eingetragen? Die wird ja in Wh und nicht in kWh angegeben.

DavidPasch, 2022-03-18:
Herzlichen Dank an euch beiden!

1) Die Idee der Programmierung ist eine Grundidee die ich ebenfalls hatte, habe ich mal im Hinterkopf. Zumal ich mir was zur dynamischen Steuerung einer Wallbox bzgl. Solar-optimiertem Laden bauen wollte.

2) Wh statt kWh ... kann die Lösung so einfach sein? Tatsächlich "überlesen" und nun richtig eingestellt. Ich beobachte das mal.

NielsWL, 2022-03-18:
Und ich werde mal die im Adapter hinterlegten intelligenten Steuerungen
testen...

SvenJ76, 2022-03-30:
Ich hab ein ähnliches Problem, vielleicht könnt Ihr mir auch helfen. Mein Adapter ist durchkonfiguriert (Batterie auch in Wh eingestellt), aber er macht mir das dynamische Batterymgmt. ständig aus. Meldung dazu ist "Disabling battery management because 40 is too high for current SoC 40". Das macht er halt auch bei strahlendem Sonnenschein wo ich mir ja wünschen würde, dass er das dynamische Mgmt einschaltet damit die Battery später lädt. 
Habt Ihr eine Idee?
[image]
?

StrathCole, 2022-03-30:
> Meldung dazu ist "Disabling battery management because 40 is too high for current SoC 40".

Der Adapter will sicherstellen, dass die Batterie nicht aus dem Netz auf das Minimum geladen wird. Daher wird das Management erst dann aktiviert, wenn die Ladung der Batterie > MinSoC + 8% ist.
In deinem Fall stand der MinSoC (seitens KOSTAL) wohl bereits auf 40%, was extrem hoch ist, und dann wird das Management erst ab einer Ladung von 48% aktiv.

SvenJ76, 2022-03-30:
ah, ok... auf was würdest Du den Wert sinnvollerweise setzen?

StrathCole, 2022-03-30:
> ah, ok... auf was würdest Du den Wert sinnvollerweise setzen?

Da du die Dynamik aktiv hast, würde ich ihn denke ich auf 15 bis maximal 20 setzen je nach Batteriegröße.

SvenJ76, 2022-03-30:
Danke für Deine Hilfe, ich hab jetzt mal auf 15 gesetzt und beobachte!

closed 2022-03-30 by StrathCole