pixcept/ioBroker.plenticore #60: Dynamischer MinSoC immer aktiv

ninja-warrior, 2022-04-26:
Hallo zusammen,

seit dem letzten Update des Adapters (v2.2.1) habe ich ein seltsames Verhalten festgestellt. Ich betreibe den Adapter mit einem Kostal Plenticore Plus 5.5 und einem BYD Batteriespeicher. Ich habe den MinSoC in der Weboberfläche des Plenticore Wechselrichters statisch auf 10 % festgelegt. Der dynamische MinSoC ist in der Konfiguration des Plenticore-Adapters und auch im Plenticore Wechselrichter selbst ausgeschaltet.

Etwa einen Tag nach dem Update habe ich festgestellt, dass sich plötzlich der MinSoC von 10 % auf 40 % geändert wurde, obwohl ich diesen nicht manuell gesetzt habe. Daraufhin habe ich im Plenticore-Adapter zu Testzwecken den min. MinSoc auf 9 % und den max. MinSoc auf 11 % eingestellt. Den dynamischen MinSoc habe ich nach wie vor deaktiviert.

Heute Nacht wurde der MinSoc ohne händisches Eingreifen wie von Zauberhand auf 11 % erhöht. Ich vermute deshalb, dass im Plenticore-Adapter der dynamische MinSoc auch dann angewendet wird, wenn dieser in der Konfiguration deaktiviert wurde.

Kann sich das mal jemand anschauen? 

Vielen Dank vorab und viele Grüße!

ninja-warrior, 2022-05-08:
Push. Immer noch relevant. Kann sich das mal jemand anschauen? Braucht ihr dazu weiteren Input? Viele Grüße!

Diginix, 2022-05-17:
Hatte gestern 30 min nach dem Update auf 2.2.2 ein ähnliches Verhalten. Aber seit dem ist es nicht wieder aufgetreten.

Diginix, 2022-05-27:
Auch bei mit wir der minSOC am Abend bei schlechten Prognosewerten für den Folgetag auf den in der Instanz konfigurierten maximalen Wert gesetzt obwohl die dynamisch minSOC Steuerung gar nicht aktiviert ist.

ninja-warrior, 2022-05-27:
Jetzt brauchen wir nur noch jemanden der sich das Thema mal anschaut. Oder zumindest eine Rückmeldung gibt.

PastCoder, 2022-05-27:
Ich kann am Sonntag mal eine Version mit ein paar Testausgaben machen, um das Thema einzugrenzen. Danach werde ich aber erst in 2-3 Wochen rein schauen können.

PastCoder, 2022-05-29:
Unter https://github.com/PastCoder/ioBroker.plenticore/tarball/investigateMinSoc ist jetzt eine Version mit ein paar wenigen Info-Aufgaben mehr. Diese sind alle beginnend mit "#TEST" gekennzeichnet. Die meisten davon gab es vorher auch, aber bei Level Debug kommt halt irre viele Meldungen.
Dieser Zweig ist nur temporär, um das oben beschrieben Verhalten einzugrenzen. Die Version basiert auf der 2.2.2.
Wenn es bei euch wieder auftritt, bitte posten, welche Meldungen mit "#TEST" im Log sind. Mal sehen, ob das schon hilft zum Eingrenzen oder mehr Ausgaben nötig sind.

Diginix, 2022-05-29:
Installiert und läuft mit Loglevel info. Da die Prognose für morgen nicht gut ist, wird er heute sicher den minSOC von 5% wieder anheben wollen. Mein max. minSOC ist in der Instanz auf 6 gestellt, damit er nicht unnötig hoch geht. 6% hatte der Adapter gestern Abend dann auch wieder gesetzt. Komischerweise kam nach dem Neustart des Adapters diese Logzeile:
`plenticore.0	2022-05-29 12:42:44.121	info	##TEST## Setting minSoc. ID: devices.local.battery.MinSocDummy curMinSoC: 5 minSoC: 5`
Obwohl der minSOC auf 6 % steht. Habe ihn danach dann auf 5 % gesetzt, damit der Adapter heute Abend hoffentlich wieder fälschlicherweise regelt.

PastCoder, 2022-05-29:
Wenn das Setzen des minSoc disabled ist, wird statt der "echten" Wertes der mit "... Dummy" gesetzt, wenn ich den bestehenden Code richtig gelesen habe. Die Ausgabe wäre also noch wie erwartet. Ich bin tatsächlich gespannt, ob der minSoc vom Adapter gesetzt wird oder ob das Verhalten durch das Update von Kostal kommt und nur zufällig zeitlich etwa mit der Version 2.2.1 zusammengefallen ist. In ein paar Tagen sehen wir mehr.

Diginix, 2022-05-29:
Kostal kann ich ausschließen weil der neue minSOC Wert immer der in der Instanz hinterlegte max minSOC war.
Hatte ich dort 20 % drin, wurde auf 20 % geregelt. Nun wo ich 6 % in der Instanz gesetzt habe, geht er nur auf 6 %.
Wäre schon seeehr komisch wenn die Wechselrichter Firmware davon was wüsste ;-)

Diginix, 2022-05-29:
Hier sieht man auch am ack=false, dass der Wert vom Adapter gesetzt wurde und als der WR ihn angenommen hat, wurde er bestätigt noch mal geschrieben:

[image]

PastCoder, 2022-05-29:
Ich habe gerade noch einen Parameter mehr in den Log-Ausgaben aufgenommen, um eine Vermutung zu bestätigen widerlegen.
Wäre nett, wenn Ihr nochmal auf die neue Version unter https://github.com/PastCoder/ioBroker.plenticore/tarball/investigateMinSoc gehen könntet.

PastCoder, 2022-05-29:
Hier noch eine Alternative: 
Da ich heute mehr Zeit hatte als gedacht (und mir der Hinweis zu 23Uhr geholfen hat), habe ich vermutlich die Ursache gefunden.
Bitte mal testen mit https://github.com/PastCoder/ioBroker.plenticore/tarball/fixMinSoc

Wenn das Problem damit gelöst sein sollte, wäre es auch gut, wenn ich es behoben habe, da ich es wohl auch in Commit e3707d2 verursacht habe.

Diginix, 2022-05-29:
Ich teste und berichte.

PastCoder, 2022-05-31:
@Diginix , @ninja-warrior 
Wenn die Änderung funktioniert, bitte Info an @StrathCole, damit er sie übernehmen kann.

Diginix, 2022-06-01:
Bei mir sieht es mit [fixMinSoc](https://github.com/PastCoder/ioBroker.plenticore/tarball/fixMinSoc) gut aus. Ertragsprognosen vom Adapter waren auch niedrig genug um regeln zu wollen. Was man an MinSocDummy sieht: 
[image]

@StrathCole Bitte die Änderungen von @PastCoder übernehmen.

closed 2022-06-01 by StrathCole