pixcept/ioBroker.plenticore #63: plenticore.0.devices.local.generator.ShadowMgmt has to be type "boolean" but received type "number"

Diginix, 2022-05-16:
```
2022-05-16 21:12:59.574 - info: plenticore.0 (351740) State value to set for "plenticore.0.devices.local.generator.ShadowMgmt" has to be type "boolean" but received type "number" 
2022-05-16 21:13:21.067 - info: plenticore.0 (351740) State value to set for "plenticore.0.devices.local.generator.ShadowMgmt" has to be type "boolean" but received type "number" 
2022-05-16 21:13:42.106 - info: plenticore.0 (351740) State value to set for "plenticore.0.devices.local.generator.ShadowMgmt" has to be type "boolean" but received type "number" 
2022-05-16 21:14:03.173 - info: plenticore.0 (351740) State value to set for "plenticore.0.devices.local.generator.ShadowMgmt" has to be type "boolean" but received type "number" 

```

adapter 2.2.2
nodejs 14.19.2
js-controller 4.0.23

StrathCole, 2022-05-16:
Hast du einmal versucht, den Datenpunkt zu löschen und den Adapter neu zu starten? Evtl. ist das noch übrig aus alten Versionen.

Diginix, 2022-05-16:
Nach manuellem Löschen des Objektes wurde es mit neuem Typ und div. anderen DIngen neu erzeugt. Passt nun.

closed 2022-05-16 by Diginix