pixcept/ioBroker.plenticore #66: Batteriewert SoC aktualisiert nicht mehr in 2.2.0

Sofasportler, 2022-06-24:
Hallo,
der Adapter lief bisher problemlos, aber seit gestern aktualisiert er den SoC Wert für die Batterie nicht mehr.
Nutze den Adapter mit einem Plenticore Plus und einer BYD Batterie.

xonan82, 2022-07-03:
Ich habe das selbe Problem, bei mir ist zum einen der SoC der Batterie und die Power bei PV2 betroffen. In der Web GUI vom Plenticore werden die Daten korrekt angegeben. Bei mir ebenfalls mit BYD Batterie.

Sofasportler, 2022-07-03:
Ich konnte das Problem lösen, in drm ich den Adapter ca 10 Minuten gestoppt habe.
Danach ging es wieder.
Aktuell läuft es.

xonan82, 2022-07-03:
Hi, 
danke für den Tipp, nach dem beschriebenen Vorgehen werden auch bei mir die Werte wieder korrekt angezeigt.

PastCoder, 2022-09-25:
@StrathCole I think this issue can be closed.

Sofasportler, 2022-09-25:
Yes, it can.