pixcept/ioBroker.plenticore #67: Glätten der Werte in plenticore.0.forecast.day1.power.day_adjusted

RePa02, 2022-08-24:
Ich nutze den Wert, um zu entscheiden ob ich noch Strom verschwenden darf oder nicht, Allerdings ergeben sich innerhalb einer Stunde mehrere Schwankungen die meines Erachtens auf einen unglücklichen Rechenweg hindeuten. Lässt sich das vielleicht glätten?
Ich kann mir die Werte zwar auch selbst weg schreiben und glätten, vermute aber das das so nicht gewollt ist.
 
[image]

Vielen Dank.

PastCoder, 2022-08-24:
Hallo,
welche Version verwendest du? Ist die Änderung von pull request #57 https://github.com/StrathCole/ioBroker.plenticore/pull/57 bereits dabei? Falls nicht, reduziert diese die Schwankung vielleicht schon genug. Ist dann auch nicht ganz weg, aber recht gering.

RePa02, 2022-08-24:
Ich habe Version 2.2.0 des Adapters. Also ist die Änderung wohl noch nicht eingeflossen. 
Sorry, habe ich nicht gesehen. :( Danke.

PastCoder, 2022-08-24:
Gerne. Kurze Rückmeldung willkommen, ob das Problem damit gelöst ist.

RePa02, 2022-08-25:
Ich hab das nun etwas laufen lassen und kann sagen das Problem ist durch die Version 2.2.2 des Adapters gelöst. Die Werte sind weitestgehend geglättet. Super!

closed 2022-08-25 by RePa02