pixcept/ioBroker.plenticore #71: Batterieladen über AC erzwingen

arne0075, 2022-12-25:
Hallo zusammen, würde mich über eine Funktion im Adapter freuen, mit der man das Laden des Hausspeichers über AC (Netz) erzwingen kann z.B. bei niedrigen Strompreisen (awattar).
VG und besten Dank

JohannesB1984, 2023-01-19:
Hallo zusammen, daran habe ich auch Interesse. Wenn man sich als Installateur anmeldet gibt es unter Batterie inzwischen einen Button Batterie auf 100% aufladen. Könnten man diesen in den Adapter einpflegen?

koeperl, 2023-02-02:
Hallo zusammen, auch ich hätte Interesse an der Funktion. Es gibt wohl einen API Befehl Battery:ExternControl:DcPowerAbs über den man die Ladeleistung in Watt einstellen kann. Wenn die vorgegebene DC-Leistung nicht vorhanden ist wird diese durch AC-Leistung ersetzt

Jey-Cee, 2023-02-27:
Bedankt euch bei arne0075, er hat mich dafür Bezahlt die Funktion im Adapter zu implementieren.
Installieren könnt ihr ihn momentan über Github von hier: https://github.com/Jey-Cee/ioBroker.plenticore bis der Maintainer vom Adapter den Pull Request #72  angenommen hat.

Thommy05, 2023-02-27:
Hallo Zusammen!
Könnte man ggf. den Vorschlag von JohannesB1984 auch mit einbauen?
Geht das über die API.
Man müsste wohl in den Einstellungen des Adapters seine Installteur-Kennung eingeben können.
Danke!
VG
Thomas

Jey-Cee, 2023-03-05:
> Hallo Zusammen! Könnte man ggf. den Vorschlag von JohannesB1984 auch mit einbauen? Geht das über die API. Man müsste wohl in den Einstellungen des Adapters seine Installteur-Kennung eingeben können. Danke! VG Thomas

Ich denke schon das man das Einbauen könnte, Grundsätzlich wäre ich auch bereit mir das an zu schauen. 
Wenn du mich damit beauftragen möchtest kannst du mir eine e-mail Schreiben: jey-cee@live.com

TobiasBraeutigam, 2023-04-21:
Danke für das tolle Feature!

@Jey-Cee
Die neuen Objekte ExternControl_DcPowerAbs und ExternControl_MaxChargePowerAbs, funktionieren nur während ExternControl der Batterie auf 2 (Modbus TCP) eingestellt ist. Die Batteriesteuerung kann aber nur über das Webinterface als Installateur auf Modbus eingestellt werden.

Der ExternControl-Wert muss dann alle 3 Minuten aktualisiert werden, andernfalls schaltet der Plenticore wieder auf interne Steuerung um, bis er einen neuen Wert erhält. 

Ist es also erforderlich, dass nach 3 Minuten ohne neuen ExternControl-Wert erneut ein Installateur die Batteriesteuerung wieder auf Modbus umstellt oder reicht es aus wenn neue ExternControl-Werte gesendet werden?

Danke und viele Grüße,
Tobias

Jey-Cee, 2023-04-22:
@thoblank Danke für den Hinweis.

@TobiasBraeutigam Die Einstellung hat ja erstmal nichts mit dem Aktuellen Betriebszustand zu tun. Sobald wieder von extern gesteuert wird wechselt der Betriebszustand. 
Die Einstellung sorgt nur dafür das Überhaupt eine Steuerung möglich ist.

TobiasBraeutigam, 2023-06-10:
Wenn ich den **ExternControl_DcPowerAbs** direkt per **Kostal API** setze, dann hat dies den gewünschten Effekt und die Batterie kann auch per AC geladen werden. Aber auch nach 3 Minuten und länger ohne weitere Eingaben stellt sich das nicht wieder in den ursprünglichen Betriebszustand zurück. 

Wird das in iOBroker.plenticore programmatisch sichergestellt? Oder liegt es an der aktuellen Plenticore UI Version 01.27.09932 ? Würde mich über Erfahrungen sehr freuen.

TobiasBraeutigam, 2023-07-06:
Hier wird Dir geholfen: https://www.photovoltaikforum.com/thread/139927-solateur-mit-kostal-service-code-gesucht/

JohannesB1984, 2023-07-30:
könnte man nicht einfach diesen Button über API verfügbar machen?
[image]

Rabbat2020, 2023-10-18:
Ich schließe mich @JohannesB1984  an.  Wäre dies möglich @Jey-Cee ? 

Ich würde gerne die Batterie zum besten günstigsten Preis (per Tibber) auf 100% aufladen .

Ich bin im Besitz des Masterkey & ServiceCode als Installateur . Ich würde mich also zum testen zu Verfügung stellen.

atonimo, 2023-10-18:
> Ich schließe mich @JohannesB1984 an. Wäre dies möglich @Jey-Cee ?
> 
> Ich würde gerne die Batterie zum besten günstigsten Preis (per Tibber) auf 100% aufladen .
> 
> Ich bin im Besitz des Masterkey & ServiceCode als Installateur . Ich würde mich also zum testen zu Verfügung stellen.

Den Be- und Entladewirkungsgrad des Akkus berücksichtigen bei der Rechnung.

Thommy05, 2023-10-18:
Jey-Cer möchte dafür ein paar Euros haben. War mit zu teuer da unwirtschaftlich.
Bin aber gerne dabei da ein paar Euro mitzuzahlen, wenn was zusammen geht. 
VG

Rabbat2020, 2023-10-18:
Ok , Kostal-API / sowie der jetzige Kostal-API-Adapter für ioBroker ist ein für ein Open-Source-Projekt, das auf den Prinzipien der Kollaboration, Gemeinschaft und offenen Quellcodes basiert .

Welche Philosophie steckt denn dahinter dieses Projekt dann hier zu erstellen und zu veröffentlichen? 
Open-Source-Softwareprojekte profitieren oft erheblich von der Mitwirkung und Zusammenarbeit der Community
VG

atonimo, 2023-10-18:
> > > Ich schließe mich @JohannesB1984 an. Wäre dies möglich @Jey-Cee ?
> > > Ich würde gerne die Batterie zum besten günstigsten Preis (per Tibber) auf 100% aufladen .
> > > Ich bin im Besitz des Masterkey & ServiceCode als Installateur . Ich würde mich also zum testen zu Verfügung stellen.
> > 
> > 
> > Den Be- und Entladewirkungsgrad des Akkus berücksichtigen bei der Rechnung.
> 
> Das macht der Plenticore doch selbst. Es soll nur der Befehl zum Aufladen gegeben werden. Oder meinst du was anderes?

Ich meine in der Berechnung der Kosten. Es bringt nichts wenn du dir beim Strompreis10% sparst aber das laden nur einen Wirkungsgrad von 80% hast.

Rabbat2020, 2023-10-18:
#
/

> Das macht der Plenticore doch selbst. Es soll nur der Befehl zum Aufladen gegeben werden. Oder meinst du was anderes?

Ja , das ist Richtig, dass der Plenticore dies selbst macht. Wenn man diesen Befehl (Button) (100% Aufladung) ausführt. Könnte man sich den Part (plenticore.X.devices.local.battery.ExternControl_DcPowerAbs) ersparen. Da dies automatisiert läuft ohne das externe Batteriemanagement zu aktivieren.

Zum Thema Kosten: Dies wäre nur für die Wintertage in Kombination mit Tibber. Die Historie Zeigt das zu den Stoßzeiten 7-9Uhr und Abends zw. 18-21Uhr die höchsten Preise sind. In der Zeit möchte man die Akku-Kappa nutzen um nicht 50cent zu zahlen oder mehr. Tibbert deckelt nur den Durchschnittlichen Strompreis am Tag auf 40cent. Somit kann passieren das du zu den Stoßzeiten 40Cent oder auch mehr zahlen musst. Und leider verbrauche ich da auch Teilweise am meisten.

Wirkungsgrad liegt bei AC/DC Wandlung bei ca. 90% / Verlust max. 10%. Kostal ist da um einiges besser.

Mein Theorie: Ganz rudimentär
100 Tage a 10KWh Akku zu 20cent Ersparnis gegenüber den Stoßzeiten = 200€ pro Winter
VG

Rabbat2020, 2023-10-18:
[image]

Nur mal für morgen

Jey-Cee, 2023-10-18:
@atonimo  ah ok der Kontext hat mir gefehlt.

JohannesB1984, 2023-10-19:
schön das wieder Leben im Post ist und viele das gleiche vorhaben. Mir würde es wirklich reichen diesen Button über den iobroker auszulösen. Die Akkuladung ist dann ja auch in 2h bei meinen 11kwh durch...
Um welchen Betrag geht es denn der dem Entwickler zukommen soll?

pixcept, 2023-10-19:
Okay, also ich hätte da zwei Sachen auf der Todo-Liste:

1. Batterieentladung sperren
Das könnte über die "Zeitgesteuerte Nutzung" gemacht werden, in dem ich dort einfach bei Aktivierung für alle Tage die Einstellung setze

2. Ladung erzwingen
Das Problem mit dem Button für die 100% Batterieladung ist, dass es eine Installateurs-Einstellung ist und somit über den Login, der vom Adapter genutzt wird, gar nicht verfügbar sein wird.

Rabbat2020, 2023-10-19:
Hallo,
für die , die einen Service Code haben, können erstmal Selenium nutzen. 
Somit könnt ihr eure Anmeldung und Button drücken automatisieren und ausführen . 
In Firefox/Chrome gibts auch ein Plugin somit könnt ihr erstmal ein das Script "recorden". 
Anschließend kann es dann auch mit Phyton ohne den Browser zu nutzen ausgeführt werden.

Nice wäre dies aber schon , wenn dies Jee-Cee API Adapter das unterstützt :-)
VG

JohannesB1984, 2023-10-19:
> Okay, also ich hätte da zwei Sachen auf der Todo-Liste:
>
> 1. Batterieentladung sperren
>    Das könnte über die "Zeitgesteuerte Nutzung" gemacht werden, in dem ich dort einfach bei Aktivierung für alle [...]

Punkt 2, habe ich gelöst beim Thema Sofortladen der Wallbox ohne Batterienutzung, das ich den Wert plenticore.0.devices.local.battery.MinHomeConsumption mit dem aktuellen Leistungswert der Wallbox aktualisiere.
Gleiches würde ich auch so machen wenn ich die Batterie aus dem Netz laden möchte, wenn Tibber bei 0 Euro liegt. Dann würde ich den Wert einfach auf 10000 setzen und die Batterie wird nicht entladen....

JohannesB1984, 2023-10-19:
> Selenium

Okay, das probiere ich mal aus. Den Code habe ich ja auch zur Hand...

PastCoder, 2023-10-23:
Ich knobel gerade an dem "kleinen Bruder" von der Frage. 
Da ich keinen Installateurscode habe, kann ich wohl nicht das Laden per AC erzwingen. 
Aber zumindest müsste ich ja das Entladen verhindern können, um die Ladung in "teuren" Tibber-Zeiten zu nutzen. 
Meine Idee wäre dafür den minSoc auf 100% zu setzen. Dann würde zumindest kein Batterie-Strom mehr verwendet. 
Gibt es bessere Möglichkeiten, um z.B. auch zu erzwingen, dass der PV-Strom zum Laden anstatt für den Hausverbrauch verwendet wird?

Für die Umsetzung wäre mein Vorschlag, ein noch recht neue Einstellung plenticore.0.devices.local.battery.SmartBatteryControlMode um weitere Modi zu erweitern.
Beispielsweise:
- load: Smart battery management ist aus
- Kostal smart: Das Plenticore smart Management ist eingeschaltet 
- auto decision: Die Logik im Plenticore-Adapter entscheidet über smart on/off und minSoc (also das normale Verhalten des Adapters)
- Keep battery load: Nutze PV zum Verbrauch aber nicht die Batterieladung
- AC Load: Aus dem Netz laden

Was meint Ihr?

Generell hatte ich die Logik zum Suchen nach den besten Zeiten für Nutzung des Speichers und Einplanen der Wärmepumpe unterschätzt wegen der vielen variablen und von einander abhängigen Elemente. Aber so langsam scheint es was sinnvolles zu tun. Jetzt muss ich es nur noch dem Adapter beibringen.

PastCoder, 2023-11-02:
Ich habe wie oben beschrieben angefangen die Option "devices.local.battery.SmartBatteryControlMode" zu erweitern, so dass es die folgenden Modi gibt: 
1: 'auto load decision',
2: 'off (load battery)',
3: 'on (Kostal smart)',
4: 'retain battery',
5: 'load from AC'

Aktuell ist es noch WorkInProgress und bitte noch nicht ausprobieren. Ich gebe Bescheid, wenn es so weit ist, dass weitere Testnutzer willkommen sind. 

Nun zu meiner Frage: 
Damit ich an Option 5 arbeiten kann, muss bei meinem Wechselrichter noch die Option zum Laden per AC freigeschaltet werden. Wenn von Euch diese Einstellung per Teamviewer und Code bei mir ändern kann, bitte kurze Nachricht. 

Für Option 4 verwende ich den Vorschlag von @JohannesB1984  und gehen über battery.MinHomeConsumption.

arne0075, 2023-11-02:
Hi zusammen, also das Batterieladen über AC oder die Sperrung der Entladung funktioniert doch bereits heute ohne weitere Anpassungen bzw. Installateurzugang. Nachfolgend mein Blockly hierfür. Wenn die Entladung gesperrt werden soll, einfach statt der -5000 eine 0 senden.
[image]

PastCoder, 2023-11-02:
@arne0075 Ich dachte, dass der Wert von 
plenticore.0.devices.local.battery.ExternControl_DcPowerAbs 
nur übernommen wird, wenn 
plenticore.0.devices.local.battery.ExternControl 
auf "External (Modbus TCP)" gesetzt ist und dass dies nur mit Installateur-Code möglich ist.

Oder war dies Verständnis falsch?

arne0075, 2023-11-02:
ne man muss nicht auf ExternalControle umstellen, das macht der Kostal automatisch solange alle mindestens 3 min der Befehl für DC power geschickt wird. Sobald kein DC Power Befehl mehr gesendet wird, geht der Kostal automatisch nach 3 min wieder in seinen Automatikmodus.

PastCoder, 2023-11-02:
> Vielen Dank schonmal für Deine Arbeit, @PastCoder Bekommt der Punkt 5 noch einen Wert übergeben, mit dem die Leistung eingestellt werden kann? Jeder Akku hat ja andere Leistungswerte (und die Anwender unterschiedliche Vorlieben).

Ist mir gerade beim Nachdenken auch aufgefallen. 
Das AC-Laden konnte ich bisher noch nicht erproben, da bei mir die Option noch nicht aktiviert ist (s.o. die Suche nach jemand, der es für mich aktivieren kann). 
Ich überlege mal, wie man das so lösen kann, dass es "geradlinig" nutzbar ist (also im Idealfall nicht mehrere von einander abhängige Parameter hat). 
Wäre es eher eine Einstellung, die man einmalig machen will oder eher variabel (wenn z.B. das imaginäre Windrad N Watt AC übrig hat, will ich mit N Watt laden)?

PastCoder, 2023-11-03:
Hier der Vorschlag zur Umsetzung.
Ist zwar etwas mehr Änderung als ich mir erst vorgenommen hatte, aber ich habe bisher keine bessere Idee, die nicht irgendwann in sehr verknoteten Einstellungen endet.
**Feedback willkommen -** vor allem auch, weil es vermutlich nur als "breaking change" ginge (also nicht ohne kleine Interaktion des Benutzers).

- Verschmelzen der bisherigen zwei Einstellungen in einen neuen `ChargeControlMode `
    - `plenticore.0.devices.local.battery.SmartBatteryControlMode`
    - "Select battery strategy" from instance settings page
- Einstellungen, die das Verhalten des Adapters beeinflussen trennen von den eigentlichen Einstellungen des Plenticore. Hier einen neuen Zweig: `plenticore.0.behavior`.

Neue Einstellungen (Verwendung siehe unten):
- `plenticore.0.behavior.ChargeControlMode` (default 0)
- `plenticore.0.behavior.AcLoadPower` (default 0)
- `plenticore.0.behavior.MinHomeConsumptionNormal` (default 50)
- `plenticore.0.behavior.MinHomeConsumptionRetainBattery` (default 10000)
If possible, set default values from current setting in plenticore when adapter is installed or started the first time.
Another option would be to use default "-1" for all settings and not to change the related Plenticore settings as long as the value is -1. This would prevent the adapter from changing something which was not intended by the user.

Verhalten der neuen Einstellung:
`plenticore.0.devices.local.behavior.ChargeControlMode` with following options:
- 0: 'off (load battery)',
- 1: 'on (Kostal smart)',
- 2: 'retain battery',
- 3: 'load from AC'
- (reseved for future modes)
- 11: 'Strategy 1',
- 12: 'Strategy 2'
- (reserved for future strategies)

Precondition for mode '3 load from AC' is that `plenticore.0.devices.local.battery.ExternControl` is set to "External (Modbus TCP)". This setting can be change only with Kostal installer code.
For mode '2 retain battery' the approach with MinHomeConsumption is used due to the following reasons: 
According to forum information the change of MinSoc causes sometimes problems; and the MinSoc would need to be reset to the earlier value when switching to another mode.
The option via ExternControl_DcPowerAbs is available only to users where be the beformentioned precondition is met.

Beeinflusste Plenticore-Einstellungen:

- 0: 'off (load battery)'
`plenticore.0.devices.local.battery.SmartBatteryControl` = false
`plenticore.0.devices.local.battery.MinHomeConsumption` = _MinHomeConsumptionNormal_
`plenticore.0.devices.local.battery.ExternControl_DcPowerAbs` = 0
- 1: 'on (Kostal smart)',
`plenticore.0.devices.local.battery.SmartBatteryControl` = true
`plenticore.0.devices.local.battery.MinHomeConsumption` = _MinHomeConsumptionNormal_
`plenticore.0.devices.local.battery.ExternControl_DcPowerAbs` = 0
- 2: 'retain battery',
`plenticore.0.devices.local.battery.SmartBatteryControl` = false
`plenticore.0.devices.local.battery.MinHomeConsumption` = _MinHomeConsumptionRetainBattery_
`plenticore.0.devices.local.battery.ExternControl_DcPowerAbs` = 0
- 3: 'load from AC'
`plenticore.0.devices.local.battery.SmartBatteryControl` = false
`plenticore.0.devices.local.battery.MinHomeConsumption` = _MinHomeConsumptionNormal_
`plenticore.0.devices.local.battery.ExternControl_DcPowerAbs` = _AcLoadPower_
- 11: 'Strategy 1',
`plenticore.0.devices.local.battery.SmartBatteryControl` = _decided by strategy_
`plenticore.0.devices.local.battery.MinHomeConsumption` = _MinHomeConsumptionNormal_
`plenticore.0.devices.local.battery.ExternControl_DcPowerAbs` = 0
- 12: 'Strategy 2'
`plenticore.0.devices.local.battery.SmartBatteryControl` = _decided by strategy_
`plenticore.0.devices.local.battery.MinHomeConsumption` = _MinHomeConsumptionNormal_
`plenticore.0.devices.local.battery.ExternControl_DcPowerAbs` = 0

PastCoder, 2023-11-04:
> Hier der Vorschlag zur Umsetzung.

Gerade noch ein weiterer Gedank dazu: Diese Möglichkeiten werden vermutlich nur von eher technikaffinen Menschen genutzt. Und diese können dies ohne Änderung am Adapter vermutlich ohne viel Mühe mit Blockly nachbilden. 
Das würde dafür sprechen doch nichts am Adapter zu ändern.

arne0075, 2023-11-04:
> Interessanter Aufbau. Ich könnte das auch nochmal so probieren. Den Button den ich im Installationsmenü sehe hat den Nachteil, dass dieser dann den Akku voll macht unabhängig der Quelle und der Ladevorgang nicht abgebrochen werden kann.
> Was setzt Du bei Preisbasiertes laden wenn ich fragen darf, bzw. welche Logik hast Du hier angewendet und welchen Akku hast Du am Plenticore? Ich hatte immer verstanden man muss auch die Ladeleistung angeben, das fehlt ja hier im Abschnitt...

Zur Logik. Der Ziel-SOC wird am Abend in Abhängigkeit von Außentemp. und Sonnenstunden für die Nacht definiert. Dann wird berechnet wie lange der Speicher bis zum Ziel-SOC laden muss. Anschließend werden die günstigsten Stunden gewählt.

Nachfolgend das Blockly. Geht bestimmt auch eleganter..

Dran hängt ein BYD HVM 22.1. Die -5000 entspricht einer Ladeleistung von 5.000 W. Viel mehr kann der Kostal ja leider nicht.
[image]

JohannesB1984, 2023-11-05:
> Zur Logik. Der Ziel-SOC wird am Abend in Abhängigkeit von Außentemp. und Sonnenstunden für die Nacht definiert. Dann wird berechnet wie lange der Speicher bis zum Ziel-SOC laden muss. Anschließend [...]

Vom Aufbau passt es, habe auch einen BYD aber nur mit 11,7kwh am Plenticore. Hast Du das Blockly auch noch in komplett? Kann mich da bezüglich der Preisbestimmung nicht so eindenken. Was ich klasse finde ist abhängig vom Ladestand das ganze zu steuern.
So was ähnliches habe ich mit der Übeschusssteuerung und meinem Heizstab gebaut...

JohannesB1984, 2023-11-05:
Hab es mal versucht etwas nachzubauen, habe allerdings das Problem wenn ich 1080 auf Wert 2 setzte via Modbus wird dieser direkt wieder auf 0 gesetzt. Muss noch ein weiterer Wert angepasst werden? Die Einstellung 0 möchte ich ja erhalten wenn ich genug PV Strom selbst erzeuge.

topcaser, 2024-05-17:
Hallo, ich habe den Parameter ExternControlDcPowerAbs nicht bei mir verfügbar. Es ist lediglich der Parameter ExternControl vorhanden. Wie aktiviert ihr den ExternControlDcPowerAbs?

topcaser, 2024-05-17:
ok, ich habe festgestellt, dass auf meinem ioBroker nur die Version 2.2.0 läuft. Ein Update auf 2.3.1 wird mir allerdings nicht angezeigt. Warum ist das so? Kann hier jemand helfen?

BeckHilscher2024, 2024-11-22:
@topcaser um ein Update auf 2.3.1 zu machen musst Du in den Einstellungen "Repositories" von "Stable" auf "Beta" umstellen. Dann wird Dir die Version 2.3.1 angeboten