pixcept/ioBroker.plenticore #73: please merge fixMinSoc from PastCoder

Diginix, 2023-03-06:
@Jey-Cee Did you merge the @PastCoder minSOC fix from here github:PastCoder/ioBroker.plenticore#fixMinSoc?
It is missing in the last 2.2.2 from @StrathCole here. But if you continue now the development, it should be merged asap. Otherwise I have to fork my own fixed version and merge your new stuff. But this is the worst solution for all.

StrathCole, 2023-03-06:
Didn't knew someone took on development. See no pull request from the mentioned repo.

StrathCole, 2023-03-06:
Did you maybe mean this fix that was merged already:
https://github.com/StrathCole/ioBroker.plenticore/pull/65

Diginix, 2023-03-06:
This was discussed in forum and I have tried your 2.2.2 end of summer 2022 and it was not the same as the fork mention above by me. Only this from @PastCoder works without minSOC probs. It was long time ago and I can't remember without reading the thread again. I have the fixed fork running and maybe I could build a diff to 2.2.2 or 2.3.0. But means effort.

Diginix, 2023-03-06:
This suggests that it should be merged: https://github.com/StrathCole/ioBroker.plenticore/issues/60#issuecomment-1143178964
But afaik it isn't. Maybe PastCoder did another fix later.

Diginix, 2023-03-06:
Strange, here is the correct commit:
https://github.com/StrathCole/ioBroker.plenticore/commit/1cfe951db5922c123b9bf405d0ceebd7429a5e92

I'm sure that I have tested official 2.2.2 in Sep/Oct 2022 and did a rollback to PastCoder's fork on 2nd October because something was not working fine with the official version. But the fix commit was merged in May.

I will try again now with 2.3.0 and come back if I found something.

closed 2023-03-06 by Diginix

PastCoder, 2023-03-06:
If I remember correctly, the food was merged in github but that version was never released. Not sure if the version number changed. 
So maybe there were two different contents with same version number.

Diginix, 2023-03-06:
This would be an explanation!
I have installed 2.3.0 now and compared plenticore.js and the minSOC fix is included.