pixcept/ioBroker.plenticore #78: Update stable version in repo from 2.2.0 to 2.3.0

GermanBluefox, 2023-06-25:
# Think about update stable version to 2.3.0
**Version**: stable=**2.2.0** (746 days old) => latest=**2.3.0** (111 days old)
**Installs**: stable=**1244** (82.17%), latest=**157** (10.37%), total=**1514**
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L1822)

mcm1957, 2023-07-13:
A kind reminder.
Is there any reason to delay an update of stable revision?

Feel free to drop a simple note if I should trigger an update.

PastCoder, 2023-08-23:
Related to https://github.com/pixcept/ioBroker.plenticore/pull/81 I recognized that I should make the labels better understandable. I could do that at the weekend.

mcm1957, 2023-09-15:
As adapter 2.2.0 seems to fail with new js-controller 5 an update to 2.3.0 seems to be prior.
We did not get an request at ioBroker.repositories until now. Is it OK for you that we update stabel to 2.3.0 anyway?

Please let me know.
THANKS

pixcept, 2023-09-15:
Question: Either the current changes are all not going into 2.3.0 or it needs to be a 2.3.1 because 2.3.0 was already published to npm while ago.

pixcept, 2023-09-15:
So which to put in the stable repo file?

mcm1957, 2023-09-15:
Thats completly up to you. A version should go to stable after it has been available at latest for some time and the developer thinks theres not bigger problem with it.

To bump a version to stable simple use www.iobroker.dev  and open your adapter 'releases' and click at 'set stable' next to desired release.

Alternativly you can update the release menually at ioBroker.repositories within file sources-dist-stable.json.

closed 2023-10-04 by GermanBluefox