pixcept/ioBroker.plenticore #88: Please add standard guthub testing for ioBroker

mcm1957, 2023-10-17:
Please add standard testscripts and workflow test-and-release and ensure that test are passing.

You can copy the worklow from the adapter template or a newly created adapter.

If you need help setting up test-and-release please let us know. (Mention me eventually explicitly if I miss feedback here)

KoMa1978, 2024-01-11:
Would be great to have this adapter integrated into default Adapter repository

mcm1957, 2024-01-11:
> Would be great to have this adapter integrated into default Adapter repository

Adapter IS available at standard repositories.
Update of stable release requires fullfullment of minimal standards (adding standard github testing workflow). Its up to the maintainer to add this - but if course if any help is needed we will try to do so.