pixcept/ioBroker.plenticore #90: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0' after js.update

andre-bu, 2023-10-30:
Hello after updating to js.controller 5.0.12 I get this error message.
I have now updated from Versuon 2.2.0 to version 2.3.0. The problem remains
Thank you for your effort

js.controller 5.0.12
NPM 9.6.7
Node.js 18.17.1

`<html><body>

plenticore.0 | 2023-10-30 17:09:45.518 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
-- | -- | -- | --
plenticore.0 | 2023-10-30 17:09:44.525 | error | Exception-Code:  ERR_INVALID_ARG_TYPE: The "path" argument must be of type string or an  instance of Buffer or URL. Received undefined
plenticore.0 | 2023-10-30 17:09:44.523 | error | TypeError  [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string or  an instance of Buffer or URL. Received undefined     at Object.openSync (node:fs:595:10)     at Object.writeFileSync (node:fs:2324:35)     at Object.unload  (/opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:789:5)     at AdapterClass.<anonymous>  (/opt/iobroker/node_modules/iobroker.plenticore/main.js:40:14)     at AdapterClass.emit (node:events:514:28)     at AdapterClass._stop  (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:1369:26)     at AdapterClass._exceptionHandler  (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:9113:18)     at process.processTicksAndRejections  (node:internal/process/task_queues:95:5)
plenticore.0 | 2023-10-30 17:09:44.519 | error | unhandled promise rejection: The "path" argument must be of type string or an instance of Buffer or URL. Received undefined
plenticore.0 | 2023-10-30 17:09:44.518 | error | Unhandled  promise rejection. This error originated either by throwing inside of  an async function without a catch block, or by rejecting a promise which  was not handled with .catch().
plenticore.0 | 2023-10-30 17:09:44.501 | error | Exception-Code: ENOENT: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'
plenticore.0 | 2023-10-30 17:09:44.500 | error | Error: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'     at Object.mkdirSync (node:fs:1398:3)     at Object.setup (/opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:711:6)     at Immediate.<anonymous> (/opt/iobroker/node_modules/iobroker.plenticore/main.js:189:17)     at process.processImmediate (node:internal/timers:478:21)
plenticore.0 | 2023-10-30 17:09:44.497 | error | uncaught exception: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'
plenticore.0 | 2023-10-30 17:09:44.491 | info | [INFO] Configured polling interval: 20000
plenticore.0 | 2023-10-30 17:09:44.489 | info | Configured battery strategy is: 1
plenticore.0 | 2023-10-30 17:09:44.488 | info | Configured maximum feed-in power is 5000
plenticore.0 | 2023-10-30 17:09:44.486 | info | Configured Wp of panel(s) is 0
plenticore.0 | 2023-10-30 17:09:44.426 | info | [START] Starting adapter plenticore v2.3.0r237
plenticore.0 | 2023-10-30 17:09:44.348 | info | starting. Version 2.3.0 in /opt/iobroker/node_modules/iobroker.plenticore, node: v18.17.1, js-controller: 5.0.12
plenticore.0 | 2023-10-30 17:09:09.432 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
plenticore.0 | 2023-10-30 17:09:08.439 | error | Exception-Code:  ERR_INVALID_ARG_TYPE: The "path" argument must be of type string or an  instance of Buffer or URL. Received undefined
plenticore.0 | 2023-10-30 17:09:08.438 | error | TypeError  [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string or  an instance of Buffer or URL. Received undefined     at Object.openSync (node:fs:595:10)     at Object.writeFileSync (node:fs:2324:35)     at Object.unload  (/opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:789:5)     at AdapterClass.<anonymous>  (/opt/iobroker/node_modules/iobroker.plenticore/main.js:40:14)     at AdapterClass.emit (node:events:514:28)     at AdapterClass._stop  (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:1369:26)     at AdapterClass._exceptionHandler  (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:9113:18)     at process.processTicksAndRejections  (node:internal/process/task_queues:95:5)
plenticore.0 | 2023-10-30 17:09:08.432 | error | unhandled promise rejection: The "path" argument must be of type string or an instance of Buffer or URL. Received undefined
plenticore.0 | 2023-10-30 17:09:08.431 | error | Unhandled  promise rejection. This error originated either by throwing inside of  an async function without a catch block, or by rejecting a promise which  was not handled with .catch().
plenticore.0 | 2023-10-30 17:09:08.387 | error | Exception-Code: ENOENT: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'
plenticore.0 | 2023-10-30 17:09:08.385 | error | Error: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'     at Object.mkdirSync (node:fs:1398:3)     at Object.setup (/opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:711:6)     at Immediate.<anonymous> (/opt/iobroker/node_modules/iobroker.plenticore/main.js:189:17)     at process.processImmediate (node:internal/timers:478:21)
plenticore.0 | 2023-10-30 17:09:08.380 | error | uncaught exception: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'
plenticore.0 | 2023-10-30 17:09:08.373 | info | [INFO] Configured polling interval: 20000
plenticore.0 | 2023-10-30 17:09:08.370 | info | Configured battery strategy is: 1
plenticore.0 | 2023-10-30 17:09:08.369 | info | Configured maximum feed-in power is 5000
plenticore.0 | 2023-10-30 17:09:08.367 | info | Configured Wp of panel(s) is 0
plenticore.0 | 2023-10-30 17:09:08.295 | info | [START] Starting adapter plenticore v2.3.0r237
plenticore.0 | 2023-10-30 17:09:08.229 | info | starting. Version 2.3.0 in /opt/iobroker/node_modules/iobroker.plenticore, node: v18.17.1, js-controller: 5.0.12
plenticore.0 | 2023-10-30 17:08:33.385 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
plenticore.0 | 2023-10-30 17:08:32.392 | error | Exception-Code:  ERR_INVALID_ARG_TYPE: The "path" argument must be of type string or an  instance of Buffer or URL. Received undefined
plenticore.0 | 2023-10-30 17:08:32.391 | error | TypeError  [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string or  an instance of Buffer or URL. Received undefined    at Object.openSync  (node:fs:595:10)    at Object.writeFileSync (node:fs:2324:35)    at  Object.unload  (/opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:789:5)     at AdapterClass.<anonymous>  (/opt/iobroker/node_modules/iobroker.plenticore/main.js:40:14)    at  AdapterClass.emit (node:events:514:28)    at AdapterClass._stop  (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:1369:26)     at AdapterClass._exceptionHandler  (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:9113:18)     at process.processTicksAndRejections  (node:internal/process/task_queues:95:5)
plenticore.0 | 2023-10-30 17:08:32.386 | error | unhandled promise rejection: The "path" argument must be of type string or an instance of Buffer or URL. Received undefined
plenticore.0 | 2023-10-30 17:08:32.384 | error | Unhandled  promise rejection. This error originated either by throwing inside of  an async function without a catch block, or by rejecting a promise which  was not handled with .catch().
plenticore.0 | 2023-10-30 17:08:32.357 | error | Exception-Code: ENOENT: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'
plenticore.0 | 2023-10-30 17:08:32.356 | error | Error:  ENOENT: no such file or directory, mkdir  '/opt/iobroker/node_modules/iobroker-data/plenticore.0'    at  Object.mkdirSync (node:fs:1398:3)    at Object.setup  (/opt/iobroker/node_modules/iobroker.plenticore/lib/plenticore.js:711:6)     at Immediate.<anonymous>  (/opt/iobroker/node_modules/iobroker.plenticore/main.js:189:17)    at  process.processImmediate (node:internal/timers:478:21)
plenticore.0 | 2023-10-30 17:08:32.353 | error | uncaught exception: ENOENT: no such file or directory, mkdir '/opt/iobroker/node_modules/iobroker-data/plenticore.0'

</body>
</html>`

PastCoder, 2023-10-30:
I am using the following versions witout any issue:

- js-controller: 5.0.12
- Node.js: v18.18.2
- NPM: 9.8.1

With another adapter I had problems after update to 
npm 9.8.1 and node 18.18.0
This was gone after update to v18.18.1.

andre-bu, 2023-10-30:
I have now rolled back to js.controller 4.0.24. Everything runs smoothly there. As soon as I upgrade again to 5.0.12 the error message appears again.

andre-bu, 2023-10-30:
I checked with FileZilla. There is no "iobroker-data" folder in the "/opt/iobroker/node_modules" folder
@PastCoder Could you please check whether this folder is available to you?

PastCoder, 2023-10-30:
No, in my /opt/iobroker/node_modules there is no folder iobroker-data but several iobroker.* like iobroker.plenticore

PastCoder, 2023-10-30:
BTW: That's the problem I had after some updates: https://github.com/ioBroker/ioBroker.zigbee/issues/1893

mcm1957, 2023-10-31:
duplicate of https://github.com/pixcept/ioBroker.plenticore/issues/84

maintainer must update code as suggested at #84

andre-bu, 2023-10-31:
@mcm1957 However, on my test system the adapter runs without any problems with js.controller version 5.0.12

ak1de, 2023-11-04:
I do have exactly the same problem as @andre-bu after updating to js-controller 5.0.16 today.

pixcept, 2023-11-07:
Please check if the latest git version fixes it.

andre-bu, 2023-11-07:
@pixcept Many thanks for your effort. I'll try it out and give feedback.

andre-bu, 2023-11-07:
@pixcept should I use Git Url or “latest” repo

pixcept, 2023-11-07:
Git URL please.

andre-bu, 2023-11-08:
@pixcept Thank you very much, it worked.

ak1de, 2023-11-11:
> @pixcept Thank you very much, it worked.

Confirmed. Thank you!

closed 2024-01-31 by pixcept