pixcept/ioBroker.plenticore #97: Wechselwirkungen MinSOC und Ladesteuerung Wallbox (EVCC)

menkej, 2024-01-29:
Feature Request: Die Steuerung des MinSOC ist eine gute Sache. Allerdings ergibt sich jetzt eine Wechselwirkung mit EVCC. Lade ich das Auto schnell und will vermeiden, dass der Akku dabei entladen wird, setzt EVCC minSOC = aktueller SOC und damit wird der Akku nicht mehr entladen. Irgendwann funkt aber ioBroker dazwischen und setzt wieder den eingestellten/errechneten minSOC. Es wäre super, wenn dies nicht mehrmals am Tag, sondern nur 1 mal (z.B. bei Sonnnenaufgang, oder zu einem konfigurierbaren Zeitpunkt) passieren würde.

PastCoder, 2024-01-29:
Statt den MonSOC zu erhöhen müsste es eigentlich auch gehen battery.MinHomeConsumption auf einen sehr hohen Wert zu setze (z.B. 10000). Dann wird der Speicher nur entladen, wenn mehr als dieser Wert entnommen werden soll. 
Das hat laut Forum wohl weniger Nebenwirkungen als der MonSOC. Ich nutze dies auch, um Ladung im Speicher zu halten mit Zeiten hoher Strompreise (dynamischer Tarif).

PastCoder, 2024-02-26:
Ich habe gerade folgendes Verhalten festgestellt:
Während die Erzeugung der PV auf 0 war, habe ich MinHomeConsumption von 10000 auf 50 geändert. 
Das hatte so lange keine Wirkung (Es wurde also wieterhin aus den Netz Strom bezogen), bis die PV-Erzeugung begonnen hat.

Kennt jemand das Verhalten?