pixcept/ioBroker.plenticore #105: Warning for Plenticore G3, Requesting settings failed with code 503

deepdreams71, 2024-07-01:
Hi

 I get with Plenticore G3 FW 2.4.2 following warnings.

Requesting settings - [{"moduleid":"devices:local","processdataids":["Dc_P","DigitalIn","EM_State","HomeBat_P","HomeGrid_P","HomeOwn_P","HomePv_P","Home_P","Inverter:State","LimitEvuAbs"]},{"moduleid":"devices:local:ac","processdataids":["CosPhi","Frequency","L1_I","L1_P","L1_U","L2_I","L2_P","L2_U","L3_I","L3_P","L3_U","P","Q","S"]},{"moduleid":"devices:local:battery","processdataids":["Cycles","SoC","I","U","P"]},{"moduleid":"devices:local:pv1","processdataids":["I","U","P"]},{"moduleid":"devices:local:pv2","processdataids":["I","U","P"]},{"moduleid":"scb:export","processdataids":["PortalConActive"]},{"moduleid":"scb:statistic:EnergyFlow","processdataids":["Statistic:Autarky:Day","Statistic:Autarky:Month","Statistic:Autarky:Total","Statistic:Autarky:Year","Statistic:EnergyHome:Day","Statistic:EnergyHome:Month","Statistic:EnergyHome:Total","Statistic:EnergyHome:Year","Statistic:EnergyHomeBat:Day","Statistic:EnergyHomeBat:Month","Statistic:EnergyHomeBat:Total","Statistic:EnergyHomeBat:Year","Statistic:EnergyHomeGrid:Day","Statistic:EnergyHomeGrid:Month","Statistic:EnergyHomeGrid:Total","Statistic:EnergyHomeGrid:Year","Statistic:EnergyHomePv:Day","Statistic:EnergyHomePv:Month","Statistic:EnergyHomePv:Total","Statistic:EnergyHomePv:Year","Statistic:OwnConsumptionRate:Day","Statistic:OwnConsumptionRate:Month","Statistic:OwnConsumptionRate:Total","Statistic:OwnConsumptionRate:Year","Statistic:Yield:Day","Statistic:Yield:Month","Statistic:Yield:Total","Statistic:Yield:Year","Statistic:CO2Saving:Day","Statistic:CO2Saving:Month","Statistic:CO2Saving:Year","Statistic:CO2Saving:Total"]}]) failed with code 503: {"message":"internal communication error, try again later"}

Any idea?

Greetings

Olaf

mule1972, 2024-07-08:
I am getting this error too. The plenticore addon uses the unofficial API from Kostal. I just checked the API on my plenticre and it throws this error when trying to resquest the settings of the plenticore. Maybe simply a bug by Kostal or Kostal is not maintaining the API anymore and therefore they are ignoring these errors. Let`s wait for the next firmware update.

deepdreams71, 2024-07-08:
Okay, thanks, good to know

mule1972, 2024-07-10:
It is not a bug in the API as I thought at first look. The adapter requests a setting parameter (Generator:ExtModuleControl:Enable) which seems not to exists anymore in Plenticore G3 API.

One can solve this issue by removing the following code from file "plenticore.js" 
`			"Generator:ExtModuleControl:Enable": {
				id: "devices.local.generator.ExtModuleControl",
				type: 'boolean'
			},`

PastCoder, 2024-07-10:
Because someone may use it for Plenticore or Plenticore G2, it is probably better not to remove but add conditions for the device type. 
It may also not be the only change, because G3 seems to get new updates which G1 and G2 do not get.

mule1972, 2024-07-10:
My post was just meant as a workaround hint for deepdreams71. For a pull request I agree with you that one should add conditions depending on the device type.

deepdreams71, 2024-07-10:
I got the information from kostal support. that one option is not more available anymore, I can’t prove that.
 called dynamical Battery management.
If someone can compare
For G2 there should be 3 options
For G3 only 2
[image]
[image]

mule1972, 2024-07-10:
I can confirm that there are only 2 options available on G3. But i do not believe that the paramater "Generator:ExtModuleControl:Enable" which is causing the mentioned issue has anything to do with battery management.

deepdreams71, 2024-07-12:
I can confirm that the solution you offered works fine now. 

I comment out "Generator:ExtModuleControl:Enable": { id: "devices.local.generator.ExtModuleControl", type: 'boolean' },

Log looks well

[image]