pixcept/ioBroker.plenticore #114: Ständige Fehlermeldungen

JoergGiencke, 2025-01-08:
Der Adapter funktioniert, aber ich bekomme bei jeder Abfrage folgende Fehlermeldung:

`plenticore.0 | 2025-01-08 13:06:43.029 | warn | Requesting settings - [{"moduleid":"devices:local","processdataids":["Dc_P","DigitalIn","EM_State","HomeBat_P","HomeGrid_P","HomeOwn_P","HomePv_P","Home_P","Inverter:State","LimitEvuAbs"]},{"moduleid":"devices:local:ac","processdataids":["CosPhi","Frequency","L1_I","L1_P","L1_U","L2_I","L2_P","L2_U","L3_I","L3_P","L3_U","P","Q","S"]},{"moduleid":"devices:local:battery","processdataids":["Cycles","SoC","I","U","P"]},{"moduleid":"devices:local:pv1","processdataids":["I","U","P"]},{"moduleid":"devices:local:pv2","processdataids":["I","U","P"]},{"moduleid":"scb:export","processdataids":["PortalConActive"]},{"moduleid":"scb:statistic:EnergyFlow","processdataids":["Statistic:Autarky:Day","Statistic:Autarky:Month","Statistic:Autarky:Total","Statistic:Autarky:Year","Statistic:EnergyHome:Day","Statistic:EnergyHome:Month","Statistic:EnergyHome:Total","Statistic:EnergyHome:Year","Statistic:EnergyHomeBat:Day","Statistic:EnergyHomeBat:Month","Statistic:EnergyHomeBat:Total","Statistic:EnergyHomeBat:Year","Statistic:EnergyHomeGrid:Day","Statistic:EnergyHomeGrid:Month","Statistic:EnergyHomeGrid:Total","Statistic:EnergyHomeGrid:Year","Statistic:EnergyHomePv:Day","Statistic:EnergyHomePv:Month","Statistic:EnergyHomePv:Total","Statistic:EnergyHomePv:Year","Statistic:OwnConsumptionRate:Day","Statistic:OwnConsumptionRate:Month","Statistic:OwnConsumptionRate:Total","Statistic:OwnConsumptionRate:Year","Statistic:Yield:Day","Statistic:Yield:Month","Statistic:Yield:Total","Statistic:Yield:Year","Statistic:CO2Saving:Day","Statistic:CO2Saving:Month","Statistic:CO2Saving:Year","Statistic:CO2Saving:Total"]}]) failed with code 503: {"message":"internal communication error, try again later"}`

FernetMenta, 2025-02-15:
@pixcept are you still around?

pixcept, 2025-02-18:
yes.

pixcept, 2025-02-18:
A 503 error indicates "service not available". Worst case it was a change in the firmware, which prevents the access in the way the adapter does.

FernetMenta, 2025-02-18:
@pixcept the issue is cause by querying invalid setting ids. Here is a list of all settings of my plenticore G3. For example settiongs for network do no match

```
devices:local/ActivePower:ExtCtrl:ModeGradient
devices:local/ActivePower:ExtCtrl:ModeGradientEnable
devices:local/ActivePower:ExtCtrl:ModeGradientLowPriority
devices:local/ActivePower:ExtCtrl:ModePT1Enable
devices:local/ActivePower:ExtCtrl:ModePT1LowPriorityTau
devices:local/ActivePower:ExtCtrl:ModePT1Tau
devices:local/ActivePower:ExtCtrl:RampTime
devices:local/ActivePower:ExtCtrlP:LimitGridSupply
devices:local/ActivePower:ExtCtrlP:LowPriorityP
devices:local/ActivePower:ExtCtrlP:P
devices:local/ActivePower:ExtCtrlP:P_fine
devices:local/ActivePower:ExtCtrlP:Psoll
devices:local/ActivePower:SetValuePrioHighMode
devices:local/ActivePower:SetValuePrioLowMode
devices:local/Battery:ComMonitor:Enable
devices:local/Battery:ComMonitor:Time
devices:local/Battery:DynamicSoc:Enable
devices:local/Battery:Enabled
devices:local/Battery:Esb:MinSoC
devices:local/Battery:Esb:StartSoC
devices:local/Battery:ExternControl
devices:local/Battery:ExternControl:AcPowerAbs
devices:local/Battery:ExternControl:AcPowerRel
devices:local/Battery:ExternControl:DcCurrentAbs
devices:local/Battery:ExternControl:DcCurrentRel
devices:local/Battery:ExternControl:DcPowerAbs
devices:local/Battery:ExternControl:DcPowerRel
devices:local/Battery:ExternControl:MaxChargePowerAbs
devices:local/Battery:ExternControl:MaxDischargePowerAbs
devices:local/Battery:ExternControl:MaxSocRel
devices:local/Battery:ExternControl:MinSocRel
devices:local/Battery:ManualCharge
devices:local/Battery:MinGridFeedIn
devices:local/Battery:MinHomeComsumption
devices:local/Battery:MinSoc
devices:local/Battery:SmartBatteryControl:Enable
devices:local/Battery:Strategy
devices:local/Battery:SupportedTypes
devices:local/Battery:TimeControl:ConfFri
devices:local/Battery:TimeControl:ConfMon
devices:local/Battery:TimeControl:ConfSat
devices:local/Battery:TimeControl:ConfSun
devices:local/Battery:TimeControl:ConfThu
devices:local/Battery:TimeControl:ConfTue
devices:local/Battery:TimeControl:ConfWed
devices:local/Battery:TimeControl:Enable
devices:local/Battery:Type
devices:local/Battery:Winter:EndMonth
devices:local/Battery:Winter:MinSoC
devices:local/Battery:Winter:StartMonth
devices:local/Branding:CustomerArtNo
devices:local/Branding:ProductName1
devices:local/Branding:ProductName2
devices:local/Branding:Type
devices:local/ConfigurationPreset:Available:Count
devices:local/ConfigurationPreset:Available:DisplayText0
devices:local/ConfigurationPreset:Available:DisplayText1
devices:local/ConfigurationPreset:Available:DisplayText2
devices:local/ConfigurationPreset:Available:DisplayText3
devices:local/ConfigurationPreset:Available:DisplayText4
devices:local/ConfigurationPreset:Available:DisplayText5
devices:local/ConfigurationPreset:Available:DisplayText6
devices:local/ConfigurationPreset:Available:DisplayText7
devices:local/ConfigurationPreset:Available:DisplayText8
devices:local/ConfigurationPreset:Available:DisplayText9
devices:local/ConfigurationPreset:Available:DisplayText10
devices:local/ConfigurationPreset:Available:DisplayText11
[... 660 lines cut ...]
scb:network/Network:WLAN:IPv4
scb:network/Network:WLAN:Status
scb:network/Servicestate
scb:rse/Inverter:PowerCtrlBroadcast:Mode
scb:system/System:EEBusState
scb:system/System:KeepAlive
scb:system/System:Language
scb:system/System:OperatingMode
scb:system/System:OperatingMode:Cmd
scb:system/System:Reboot
scb:system/System:SetupState
scb:system/System:SubOperatingMode
scb:time/AvailableTimezones
scb:time/DateTime
scb:time/NTPservers
scb:time/NTPuse
scb:time/Timezone
scb:update/Update:CheckInterval
scb:update/Update:Disable
scb:update/Update:FieldTest
scb:update/Update:LastSearchTime
scb:update/Update:Method
scb:update/Update:NextPollingTime
scb:update/Update:NextRestartTime
```

mule1972, 2025-02-21:
Nein, der Fehler liegt nicht aus dem Settings-, sondern aus der Processdata-Request. Leider ist im Code ein Fehler in der Debug-Ausgabe, so dass anstelle des Processdata-Requests der Settings-Request im Debug-Log erscheint.

`apiCall('POST', 'settings', payload_2, function(body, code, headers) {
			if(code === 200) {
				processDataResponse(body, 'settings');
			} else {
				adapter.log.warn('Requesting settings - ' + JSON.stringify(payload) + ') failed with code ' + code + ': ' + body);
			}`

Bei der Debug-Ausgabe müsste es `stringify(payload_2)` statt `stringify(payload)` lauten.

mule1972, 2025-02-21:
Da ist definitiv kein Fehler im Sourcecode. Es scheint schlicht so zu sein, dass der Plenticore mit den letzten Updates ab und zu mit dem Error 503 antwortet. Bleibt nur ignorieren. 
Mir ist auch aufgefallen, dass die Requests an die API seit den letzten beiden Updates teils merklich länger benötigen. 
Vermutlich läuft der Plenticore teils schlicht auf einen internen Timeout und gibt dann den 503 zurück.

FernetMenta, 2025-02-23:
Mein P3 hat jedenfalls die abgefragten Settings nicht (siehe oben) und liefert deswegen 503 zurück. Habe die Sourcen mal bei mir gepatched und der Fehler ist weg. Evtl ist bei der Firmware vom P3 ja etwas anders.

mule1972, 2025-02-23:
@FernetMenta Ich kann keine Settings aus dem Debug-Log finden, welche nicht auch in Deiner Liste wären. Welche meinst Du?

FernetMenta, 2025-02-23:
z.B. die ganzen Network-Settings hier: https://github.com/pixcept/ioBroker.plenticore/blob/master/lib/plenticore.js#L565
weichen von den IDs meines posts oben ab.

FernetMenta, 2025-02-23:
Und dieses Setting aus dem Code gibt es bei mir auch nicht: https://github.com/pixcept/ioBroker.plenticore/blob/master/lib/plenticore.js#L553

FernetMenta, 2025-02-24:
@mule1972 der OP beschreibt diesen Fehler aber "bei jeder Abfrage". Ich habe den gleichen 503 Fehler bei jedem Polling-Interval im Log. Um "ab und zu" ging es hier nicht.

FernetMenta, 2025-02-24:
```

Sending post data to request: [{"moduleid":"devices:local","settingids":["Battery:DynamicSoc:Enable","Battery:MinHomeComsumption","Battery:MinSoc","Battery:SmartBatteryControl:Enable","Battery:Strategy","Battery:SupportedTypes","Battery:Type","Battery:ExternControl","Inverter:MaxApparentPower","EnergySensor:InstalledSensor","OptionKeys:StateKey0","Generator:ExtModuleControl:Enable","Generator:ShadowMgmt:Enable"]},{"moduleid":"scb:network","settingids":["Hostname","IPv4Address","IPv4Auto","IPv4DNS1","IPv4DNS2","IPv4Gateway","IPv4Subnetmask"]},{"moduleid":"scb:time","settingids":["NTPuse","NTPservers","Timezone"]},{"moduleid":"scb:modbus","settingids":["ModbusEnable","ModbusUnitId"]},{"moduleid":"scb:export","settingids":["LastExport","LastExportOk","Portal","ExportEnable"]}]
--

```

FernetMenta, 2025-02-24:
btw: die SW meines G3 ist: 3.04.01.17931
Eventuell unterscheiden sich die Settings ja zw. G2 und G3 oder zwischen den SW-Versionen.

mule1972, 2025-02-25:
> [@mule1972](https://github.com/mule1972) der OP beschreibt diesen Fehler aber "bei jeder Abfrage". Ich habe den gleichen 503 Fehler bei jedem Polling-Interval im Log. Um "ab und zu" ging es hier nicht.

Bei mir kommt es ausschließlich bei dem zuvor genannten Request zu Fehlern und hier sind alle IDs korrekt und vorhanden. "Ab und zu" war etwas unglücklich ausgedrückt. Ich wollte damit sagen, dass nicht jeder Request den 503 zurückgibt. Mal liegen 30 Sekunden dazwischen, mal auch 5 Minuten. Das zeigt auch, dass es nicht an der Abfrage fehlerhafter Settings liegt. Ich nutze im Übrigen auch einen G3 mit der identischen FW 3.04.01.17931
Kommen bei Dir denn auch Fehler bei einem anderen Request, als dem von mir genannten?

FernetMenta, 2025-02-25:
Der Adapter pollt alle hardcoded Settings mit jedem Interval in dem er auch ProcessData pollt. Nachdem das in die Hose geht, werden bei mir gar keine Settings gesynct. So ist der Adapter nicht nutzbar. Nachdem hier die Entwicklung ja mehr oder weniger tot ist, habe ich angefangen einen neuen Adapter auf Basis react v5 zu schreiben. Dort mache ich dann einen Validitätscheck der Variablen. Auch werde ich nur die Basisdaten fixt pollen und die anderen optional konfigurierbar anbieten.

FernetMenta, 2025-02-25:
@mule1972 ich kann mir nicht vorstellen, dass es bei dir anders ist, weil die IDs der Settings ja hardcoded sind. Was steht denn bei dir bei den plenrice iob Objekten unter network?

mule1972, 2025-02-25:
@FernetMenta Sorry, Sorry! Ich habe total verdrängt, dass ich mir bereits letztes Jahr einen Fork erstellt hatte und darin einige Korrekturen (insbesondere bei den Settings) vorgenommen hatte. Daher tauchen bei mir keine Fehler mehr in Verbindung mit nicht mehr aktuellen Setting-Requests auf.

mcm1957, 2025-04-07:
Fell free to check adapter ioBroker..plenticore-g3
https://github.com/FernetMenta/ioBroker.plenticore-g3